openhwfoundation/cv32e40p #2: if rs1=x0  both CSRRS and CSRRC  will not write to the CSR at all

colin4124, 2016-05-15:
In _RISCV privileged specification v1.7_  2.1 Instructions to access CSRs:

> For both CSRRS and CSRRC, if rs1=x0, then the instruction will not write to the CSR at all,
> and so shall not cause any of the side effects that might otherwise occur on a CSR write. Note
> that if rs1 specifies a register holding a zero value other than x0, the instruction will still write the
> unmodified value back to the CSR.

I found `csr_we_int` in `cs_registers.sv`, it just depend on `csr_op_i`.

colin4124, 2016-05-15:
I have a question:

> The CSRRWI, CSRRSI, and CSRRCI variants are similar to CSRRW, CSRRS, and CSRRC re-
> spectively, except they update the CSR using an XLEN-bit value obtained by zero-extending a 5-bit
> immediate (zimm[4:0]) field encoded in the rs1 field instead of a value from an integer register. If
> the zimm[4:0] field is zero, then these instructions will not write to the CSR, and shall not cause
> any of the side effects that might otherwise occur on a CSR write.

Does `CSRRWI rd, csr, 0 ` not write CSR ?

gautschimi, 2016-06-03:
Hi colin
CSRRS, CSRRC, CSRRSI and SCRRCI are for sure not writing to CSR if the immediate is 0. with CSRRWI I'm actually not sure. Actually I'm not sure if it is a typo in the specifications or if it is really meant like this.

It's a bit weird that CSRRW should update if x0 is used and CSRRWI should not update it if the immediate is 0.

colin4124, 2016-06-04:
Hi @gautschimi ,

Yes, I think so.

`cs_registers.sv` does not support that CSRRS, CSRRC are not writing to CSR if the x0 is used and CSRRSI and SCRRCI are not writing to CSR if the immediate is 0. Right? Or Just I haven't found that in source code?

silabs-PaulZ, 2020-07-09:
Hi @colin4124 , @gautschimi ,

The logic to decode a CSR read or write operation can be found in the cv32e40p_decoder.sv:
https://github.com/openhwgroup/cv32e40p/blob/63b5a87d49072b5ea0574d354128d621ee95d7f6/rtl/cv32e40p_decoder.sv#L2414-L2422

It only performs a read if the CSR instruction is a SET or CLEAR with rs=0 or zimm=0 (e.g. CSRSRS, CSRRSI, CSRRC, CSRRCI)

In the current CV32E40P implementation, a CSRRWI with zimm=0 will write to the CSR. 

The latest specification has the following (Document Version 20190608-Base-Ratified):

> The CSRRWI, CSRRSI, and CSRRCI variants are similar to CSRRW, CSRRS, and CSRRC respectively,
> except they update the CSR using an XLEN-bit value obtained by zero-extending a 5-bit
> unsigned immediate (uimm[4:0]) field encoded in the rs1 field instead of a value from an integer
> register. For CSRRSI and CSRRCI, if the uimm[4:0] field is zero, then these instructions will not
> write to the CSR, and shall not cause any of the side effects that might otherwise occur on a CSR
> write. For CSRRWI, if rd=x0, then the instruction shall not read the CSR and shall not cause any
> of the side effects that might occur on a CSR read. Both CSRRSI and CSRRCI will always read
> the CSR and cause any read side effects regardless of rd and rs1 fields.

This is also reflected in the table 9.1:
[image]

The output of the simulation:
         default mtvec                           ; mtvec = 0x1
         result of:      csrrw  rd, mtvec, x0    ; mtvec = 0x0
         result of:      csrrwi rd, mtvec,  1    ; mtvec = 0x1
         result of:      csrrs  rd, mtvec, x0    ; mtvec = 0x1
         result of:      csrrsi rd, mtvec,  0    ; mtvec = 0x1
         result of:      csrrc  rd, mtvec, x0    ; mtvec = 0x1
         result of:      csrrci rd, mtvec, x0    ; mtvec = 0x1

If all agreed that the current implementation is adhering to the specification, can you please close the ticket.

Regards,
-PaulZ

closed 2020-07-17 by colin4124