openhwfoundation/cv32e40p #6: LSU assertion fails

edcarstens, 2016-12-02:
There appears to be a problem with the 2nd assertion property in riscv_load_store_unit. I believe I have a valid case where it fails. It is failing in my simulation where the grant (data_gnt_i) occurs in the same cycle as the request (data_req_o) during a write transaction. The data_rvalid_i is asserted in the same cycle as is required for a write. Since there are no intervening cycles, the current state, CS, is still IDLE when this happens. Thus, the assertion fails.

This was a write transaction, but if grant and rvalid are allowed to be asserted in the first request cycle in a read transaction, we should see the same bogus assertion failure.

edcarstens, 2016-12-02:
In my copy of riscv_load_store_unit RTL, I changed the 2nd assertion to:

`assert property (
@(posedge clk) ((CS == IDLE) && (data_gnt_i == 1'b0)) |-> (data_rvalid_i == 1'b0) );
`

Then this assertion no longer failed in my simulation.

gautschimi, 2016-12-05:
This depends on the bus protocol. In our case it is not a valid 
transaction because:
when you do a request and get a grant, the rvalid signal is expected to 
arrive *one cycle **later*.

with this bus protocol it is not possible to receive the rvalid when the 
state machine is in an idle state, therefore this assertion is correct.

I'd suggest to delay the rvalid signal in your system such that it has 
similar behavior as in the data sheet. point 3.2 of the user manual: 
http://www.pulp-platform.org/wp-content/uploads/2016/05/user_manual.pdf

if you dont use the same protocol, the statemachine of the LSU might end 
up in a non valid state. therefore I strongly suggest to align:)

best,
michael

edcarstens, 2016-12-05:
The timing diagrams leave out the case where you explicitly do a write transaction. Could you add one? That would help me to see what is expected. The verbage should make it clear that rvalid should be delayed by at least one cycle for write transactions.

gautschimi, 2016-12-05:
in the manual it says (section 3.2):

"After receiving a grant, the memory answers with a data_rvalid_i set 
high if data_rdata_i is valid. This may happen *one or more cycles* 
after the grant has been received. Note that data_rvalid_i must also be 
set when a write was performed, although the data_rdata_i has no meaning 
in this case."

the diagrams are identical for read/write transfers. the only difference 
is that for write transactions you don't need the data_rdata_i signal, 
and for read transactions you can ignore the data_wdata_o signal.

but we can update this to make it clearer.

thanks for the information,

best,
michael

edcarstens, 2016-12-05:
Thanks, I think just one timing diagram showing only the write transaction would be helpful. I misunderstood the constraints on rvalid for the write case.

closed 2017-06-21 by gautschimi