openhwfoundation/cv32e40p #8: Instruction port RGV protocol issue

lecoadou, 2017-03-31:
### Bug condition
The RGV protocol as explained in the user manual is not respected in the following case:
* Outside memory not responding immediately (= grant signal not asserted in the same cycle as request)
* Jump instruction received

### Waveforms are attached
* wave_bug_jump_1/2:
Show two examples of the abnormal behavior at two different times.
In those waves we see the address changes from the prefetch addr (0x0084) to the jump addr 
(0x008C) (see the red cursor) without waiting for the grant. This is inconsistent with the RGV protocol explained in the user manual which indicates the address is updated only after the grant response.
Consequently, if the outside memory take into account the address sent at the beginning of the request rather than the one available during the grant response (which is theoretically authorized by the protocol), the returned data will correspond to the unwanted prefetch addr, and the valid instruction targeted by the jump will be lost.
[image]
[image]

* wave_branch_ok:
Shows there is no issue when the address jump corresponds to a conditional branch instruction.
[image]

### Investigation
In this case, the address switch is controlled by the `pc_set_i` signal of "controller.sv".
In "controller.sv" we can see a condition which have certainly been added to wait for the end of the current request before updating the address.
(refer to attached file "controller_code_1")
[image]
But as the `jr_stall_o` signal is not set to 1'b1 during the "JAL" jump, the `pc_set_i` signal is immediately asserted.
If we analyze the code where `jr_stall_o` is generated, we can see it is dependent to a "JALR" jump type but not to our failing "JAL" jump.
(refer to attached file "controller_code_2")
[image]
Do not know if it is the source of the issue.

gautschimi, 2017-04-02:
>     wave_bug_jump_2

Hi there

I looked at the waveforms and I don't see a bug or any misbehavior.  
however, the waveforms you present show a not very efficient interface 
because it stalls every second cycle. is this from pulpino or some other 
implementation? can you also attach the corresponding instruction stream 
such that I can see what the instructions are?

the waveforms show the following:
the core tries to fetch from address 0x0084 and sends a request to the 
interface, but it does not receive a grant in this cycle (for some 
reason that might be related to your implementation). in the next cycle 
the core knows that it actually doesn't need the instruction at 0x0084 
because of a taken branch. and since the grant has not yet arrived, the 
fetch unit can simply request a new address (0x0008c, branch target) 
without waiting for grant of the previous request.

the core never has to wait for a grant. as long as the request is not 
granted, the core is free to change the request and address fields as 
often as it wants. the usual case is that it leaves the request high, 
and does not change the address because it still wants to access the 
same data, but in case of a branch this is not always true.

hope this helps,
best,

Michael

lecoadou, 2017-04-03:
Hi Michael,

Thank you for your prompt response.
I attached the stream and the corresponding disassembly code.
[code.zip](https://github.com/pulp-platform/riscv/files/890228/code.zip)

> "the waveforms you present show a not very efficient interface"

I agree, but "speed" is not a constraint for my application. As the protocol of the memory controller I want to connect is a simple request/acknowledge, I added few logic to be compliant with the RISC core RGV protocol (taking as reference the ri5cy_user_manual.pdf document). I favored minimum modifications (for different reasons) over speed performance.

In the user manual I read:
> "The LSU provides a valid address in data_addr_o and sets data_req_o high... After a grant was received, the address may be changed in the next cycle by the LSU."

With this description plus the timing diagrams we logically conclude that the address is stable between the beginning of the request and the end of the Grant cycle.
Thus, capturing the address at the beginning of the request and set the Grant signal one cycle later or more should be possible.

I think the behavior you explained in your reply makes sense and we can consider there is no bug. But in this case more details should be added in the RISC core specification to highlight the possible address update without waiting for the Grant.
Your explanation (third section of your reply) is very clear and could be copy-pasted in the document (+ timing diagram to illustrate).

Best regards,
Mikael Le Coadou

gautschimi, 2017-04-03:
> Hi Michael,
>
> Thank you for your prompt response.
> I attached the stream and the corresponding disassembly code.
> code.zip <https://github.com/pulp-platform/riscv/files/890228/code.zip>
>
>     "the [...]
you should set the grant when you capture the address. it's an easy fix 
for u, just set the grant whenever you capture an address to issue a 
request. anyway, we can add this to the manual, I thought it was obvious

closed 2017-04-13 by lecoadou