openhwfoundation/cv32e40p #12: Core reset sequence and simulation

sinkswim, 2017-06-20:
I am testing the RI5CY core alone (all there is in this repo, nothing from pulpino) and have problems while simulating it with my testbench. By observing waveforms many signals appear not be initialized on reset, resulting in 'X' and the controller's FSM loops in the FIRST_FETCH state.
Is there a specific reset sequence to be sent or other special initializations on the core's inputs?
My reset sequence is  the following in an `initial` statement:
`    clk_i = 1'b0;
    rst_ni = 1'b0;      // assert reset
    clock_en_i = 1'b0;
    test_en_i = 1'b0;
    fetch_enable_i = 1'b0;
    #500ns;
    clock_en_i = 1'b1;  
    #1500ns;
    rst_ni = 1'b1;  // release reset
    #1500ns;
    fetch_enable_i = 1'b1;`

gautschimi, 2017-06-21:
hi there

hm your initial block is not clear, what is commented out and what not?
anyway, I see two reasons for the behavior you describe:

1. are all input signals defined?

2. did you initialize the registers of the register file to 0? this is 
usually done by the first instructions of the initialization code (crt0.S).

best,
Michael

sinkswim, 2017-06-21:
Thank you for your prompt response.
Do you maybe refer to https://github.com/pulp-platform/pulpino/blob/master/sw/ref/crt0.riscv.S in your second point?
In general, is there a test-bench dedicated exclusively to the core? I can't find any information regarding boot from the datasheet.

gautschimi, 2017-06-21:
yes I was referring to crt0.riscv.S.

I'm afraid there is no testbench with just the core. pulpino is already 
quite basic and does not come with a lot of overhead.

if you want to use a testbench for just the core, I suggest to have a 
look at the core interfaces, model them in your testbench, and check in 
the waveforms that all signals have a defined value. but in the end you 
end up having something like an instruction and data memory in your 
testbench, which is basically a pulpino like structure.

best,
Michael

sinkswim, 2017-06-21:
Yes, that is my approach. I managed to solve it. There was an error in converting the LSU protocol with my memories. My bad. Thanks!

closed 2017-06-21 by sinkswim