openhwfoundation/cv32e40p #15: Supported Instructions

sinkswim, 2017-07-17:
I am wondering if the `{FENCE, FENCE.I}` instructions will be implemented in future releases or if you plan to create new modules to handle them. I see no implementation details in this repo or the pulpino one. The only mentions are in the riscv_tracer.
Thank you.

davideschiavone, 2020-05-17:
FENCE and FENCE.I are now both supported. Can we close the issue? @sinkswim

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB