openhwfoundation/cv32e40p #17: A problem in "prefetch_L0_buffer.sv"

wenshh3, 2017-08-07:
In the file "prefetch_L0_buffer.sv", line 667
--------------------------------------
 if (branch_i)
        begin
          instr_req_o    = 1'b1;

          if (instr_gnt_i)
            NS = WAIT_RVALID;
          else
            NS = WAIT_GNT;
        end
        else
        begin
          instr_req_o    = 1'b1;

          if (instr_gnt_i)
            NS = WAIT_RVALID;
          else
            NS = WAIT_GNT;
        end
--------------------------------------
No matter what the state "branch_i" is, the following codes are the same, is there any mistake?

igorloi, 2018-03-23:
Hey Houston,  what's the probelem?

Silabs-ArjanB, 2020-06-15:
Hi @wenshh3 The prefetch_L0_buffer is no longer supported and has been removed all together. Could you please close this issue?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB