openhwfoundation/cv32e40p #27: Hardware loop counter does not decrement if loop end boundary spans across an uncompressed instruction

mcockrell-google, 2018-03-06:
In case the hardware loop is setup in such a way that the loop end boundary splits across a word-length [non-compressed] instruction, the loop is not executed more than once. Also the loop counter is not decremented at the end of the loop, as read from the CSR.
Chances of it happening are less, but we should document this behavior.
Loop start ----+                    +----- Loop end
               |                    |
---------------|     add     |     add     |---------
               |<-  4byte  ->|<-  4byte  ->|
               |             |      |
Another interesting observation is that, when the loop start boundary is split across a word length instruction, the subsequent re-executions of the loop decode and start the execution from the second half of the first instruction.

davideschiavone, 2018-03-08:
Dear Matt, I have the impression that this may be caused by the bug in the prefetcher buffer.

Does it still behave wrongly after the fix in 5fb7e427cf1fd3c0f5219e7a5d56750dd316c9f6 ?

Thanks
Davide

davideschiavone, 2018-10-03:
This has been confirmed to be solved

Thanks
Davide

closed 2018-10-03 by davideschiavone