openhwfoundation/cv32e40p #30: Execution hang with hardware loops and timer interrupts

mcockrell-google, 2018-03-06:
An execution hang is observed with a cross product of external timer interrupts and hardware loops. The core trace is given below. The last instruction of the hardware loop (pc = 0x349a) is lbu x24, 23(x14). When the execution resumes from the interrupt handler at that pc, the opcode executed is different [lbu x24, 7(x0)], resulting in data corruption. The corrupt data possibly propagates downstream and causes a hang in execution. Note that the handler is being entered at either 0x2074 or 0x207C depending on the type of interrupt i.e. timer A compare or timer B compare interrupt.
```
873960000    21832 0000346e 0060007b lp.starti     0x0, 0x6           
874000000    21833 00003472 0140107b lp.endi       0x0, 0x14          
874040000    21834 00003476 0070307b lp.counti     x0, 0x7            
874080000    21835 0000347a 007543b3 xor           x7, x10, x7          x7=181e1b87 x10:18196ba7  x7:00077020
874120000    21836 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
874160000    21837 00003482 00cda9a3 sw            x12, 19(x27)        x12:0000238c x27:00059e30  PA:00059e43
874920000    21856 00002074 2640006f jal           x0, 612            
874960000    21857 000022d8 34202ef3 csrrs         x29, x0, 0x342      x29=8000001d
875000000    21858 000022dc 30002ef3 csrrs         x29, x0, 0x300      x29=00001880
875160000    21862 000022e0 34102ef3 csrrs         x29, x0, 0x341      x29=00003486
875200000    21863 000022e4 1a104eb7 lui           x29, 0x1a104000     x29=1a104000
875240000    21864 000022e8 004eaf03 lw            x30, 4(x29)         x30=20000000 x29:1a104000  PA:1a104004
875280000    21865 000022ec 20000fb7 lui           x31, 0x20000000     x31=20000000
875480000    21870 000022f0 01fea623 sw            x31, 12(x29)        x31:20000000 x29:1a104000  PA:1a10400c
875520000    21871 000022f4 30200073 mret                             
875920000    21881 00003486 01868cb3 add           x25, x13, x24       x25=00000000 x13:00000000 x24:00000000
875960000    21882 0000348a 007585b3 add           x11, x11, x7        x11=3037872e x11:18196ba7  x7:181e1b87
876000000    21883 0000348e 00b5f5b3 and           x11, x11, x11       x11=3037872e x11:3037872e x11:3037872e
876040000    21884 00003492 ff8d0523 sb            x24, -22(x26)       x24:00000000 x26:0006a0f0  PA:0006a0da
876080000    21885 00003496 027563b3 rem           x7, x10, x7          x7=00000077 x10:00000077  x7:181e1b87
876320000    21891 0000349a 01774c03 lbu           x24, 23(x14)        x24=00000000 x14:00065bd0  PA:00065be7
876360000    21892 0000347a 007543b3 xor           x7, x10, x7          x7=00000000 x10:00000077  x7:00000077
876560000    21897 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
876600000    21898 00003482 00cda9a3 sw            x12, 19(x27)        x12:0000238c x27:00059e30  PA:00059e43
877120000    21911 00003486 01868cb3 add           x25, x13, x24       x25=00000000 x13:00000000 x24:00000000
877360000    21917 0000348a 007585b3 add           x11, x11, x7        x11=3037872e x11:3037872e  x7:00000000
877400000    21918 0000348e 00b5f5b3 and           x11, x11, x11       x11=3037872e x11:3037872e x11:3037872e
877440000    21919 00003492 ff8d0523 sb            x24, -22(x26)       x24:00000000 x26:0006a0f0  PA:0006a0da
877480000    21920 00003496 027563b3 rem           x7, x10, x7          x7=00000077 x10:00000077  x7:00000000
878880000    21955 0000349a 01774c03 lbu           x24, 23(x14)        x24=00000000 x14:00065bd0  PA:00065be7
878920000    21956 0000347a 007543b3 xor           x7, x10, x7          x7=00000000 x10:00000077  x7:00000077
879120000    21961 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
879160000    21962 00003482 00cda9a3 sw            x12, 19(x27)        x12:0000238c x27:00059e30  PA:00059e43
879680000    21975 00003486 01868cb3 add           x25, x13, x24       x25=00000000 x13:00000000 x24:00000000
879920000    21981 0000348a 007585b3 add           x11, x11, x7        x11=3037872e x11:3037872e  x7:00000000
879960000    21982 0000348e 00b5f5b3 and           x11, x11, x11       x11=3037872e x11:3037872e x11:3037872e
880000000    21983 00003492 ff8d0523 sb            x24, -22(x26)       x24:00000000 x26:0006a0f0  PA:0006a0da
880040000    21984 00003496 027563b3 rem           x7, x10, x7          x7=00000077 x10:00000077  x7:00000000
881480000    22020 00002074 2640006f jal           x0, 612            
881560000    22022 000022d8 34202ef3 csrrs         x29, x0, 0x342      x29=8000001d
881600000    22023 000022dc 30002ef3 csrrs         x29, x0, 0x300      x29=00001880
881760000    22027 000022e0 34102ef3 csrrs         x29, x0, 0x341      x29=0000349a
881800000    22028 000022e4 1a104eb7 lui           x29, 0x1a104000     x29=1a104000
881840000    22029 000022e8 004eaf03 lw            x30, 4(x29)         x30=20000000 x29:1a104000  PA:1a104004
881880000    22030 000022ec 20000fb7 lui           x31, 0x20000000     x31=20000000
882080000    22035 000022f0 01fea623 sw            x31, 12(x29)        x31:20000000 x29:1a104000  PA:1a10400c
882120000    22036 000022f4 30200073 mret                             
882520000    22046 0000349a 00704c03 lbu           x24, 7(x0)          x24=000000xx  PA:00000007
882560000    22047 0000347a 007543b3 xor           x7, x10, x7          x7=00000000 x10:00000077  x7:00000077
882760000    22052 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
882800000    22053 00003482 00cda9a3 sw            x12, 19(x27)        x12:0000238c x27:00059e30  PA:00059e43
883320000    22066 00003486 01868cb3 add           x25, x13, x24       x25=xxxxxxxx x13:00000000 x24:000000xx
883560000    22072 0000348a 007585b3 add           x11, x11, x7        x11=3037872e x11:3037872e  x7:00000000
883600000    22073 0000348e 00b5f5b3 and           x11, x11, x11       x11=3037872e x11:3037872e x11:3037872e
883640000    22074 00003492 ff8d0523 sb            x24, -22(x26)       x24:000000xx x26:0006a0f0  PA:0006a0da
883680000    22075 00003496 027563b3 rem           x7, x10, x7          x7=00000077 x10:00000077  x7:00000000
885080000    22110 0000349a 01774c03 lbu           x24, 23(x14)        x24=00000000 x14:00065bd0  PA:00065be7
885120000    22111 0000347a 007543b3 xor           x7, x10, x7          x7=00000000 x10:00000077  x7:00000077
885320000    22116 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
885360000    22117 00003482 00cda9a3 sw            x12, 19(x27)        x12:0000238c x27:00059e30  PA:00059e43
885880000    22130 00003486 01868cb3 add           x25, x13, x24       x25=00000000 x13:00000000 x24:00000000
886120000    22136 0000348a 007585b3 add           x11, x11, x7        x11=3037872e x11:3037872e  x7:00000000
886160000    22137 0000348e 00b5f5b3 and           x11, x11, x11       x11=3037872e x11:3037872e x11:3037872e
886200000    22138 00003492 ff8d0523 sb            x24, -22(x26)       x24:00000000 x26:0006a0f0  PA:0006a0da
886240000    22139 00003496 027563b3 rem           x7, x10, x7          x7=00000077 x10:00000077  x7:00000000
887640000    22174 0000349a 01774c03 lbu           x24, 23(x14)        x24=00000000 x14:00065bd0  PA:00065be7
887680000    22175 0000347a 007543b3 xor           x7, x10, x7          x7=00000000 x10:00000077  x7:00000077
887880000    22180 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
[... 6 lines cut ...]
889000000    22208 000022e8 004eaf03 lw            x30, 4(x29)         x30=20000000 x29:1a104000  PA:1a104004
889040000    22209 000022ec 20000fb7 lui           x31, 0x20000000     x31=20000000
889240000    22214 000022f0 01fea623 sw            x31, 12(x29)        x31:20000000 x29:1a104000  PA:1a10400c
889280000    22215 000022f4 30200073 mret                             
889680000    22225 00003486 01868cb3 add           x25, x13, x24       x25=00000000 x13:00000000 x24:00000000
889720000    22226 0000348a 007585b3 add           x11, x11, x7        x11=3037872e x11:3037872e  x7:00000000
889760000    22227 0000348e 00b5f5b3 and           x11, x11, x11       x11=3037872e x11:3037872e x11:3037872e
889800000    22228 00003492 ff8d0523 sb            x24, -22(x26)       x24:00000000 x26:0006a0f0  PA:0006a0da
889840000    22229 00003496 027563b3 rem           x7, x10, x7          x7=00000077 x10:00000077  x7:00000000
[... cut ...]
892320000    22291 0000349a 00704c03 lbu           x24, 7(x0)          x24=000000xx  PA:00000007
892360000    22292 0000347a 007543b3 xor           x7, x10, x7          x7=00000000 x10:00000077  x7:00000077
892560000    22297 0000347e fe7a0503 lb            x10, -25(x20)       x10=00000077 x20:00030a70  PA:00030a57
892600000    22298 00003482 00cda9a3 sw            x12, 19(x27)        x12:0000238c x27:00059e30  PA:00059e43
893120000    22311 00003486 01868cb3 add           x25, x13, x24       x25=xxxxxxxx x13:00000000 x24:000000xx
8933 [The core trace abruptly stops here]
```

mcockrell-google, 2018-03-06:
A potential fix is being tested from https://github.com/pulp-platform/riscv/commit/5fb7e427cf1fd3c0f5219e7a5d56750dd316c9f6

davideschiavone, 2018-10-03:
This has been confirmed to be solved.

Thanks
Davide

closed 2018-10-03 by davideschiavone