openhwfoundation/cv32e40p #31: [RISCY CORE] Problem with executing ecall instruction when there's a halt request pending

taoliug, 2018-03-08:
I am working on testing the single step feature of RISCY core. 
Here's the test sequence:
- Instructions are being fetched and executed in the background
- Set DBG_IE[ECALL] = 1'b1, expect PC jumps to trap address(0x88) when ecall is executed.
- Write 'h10000 to DBG_CTRL from debug interface to enter the debug mode
- There's a delay between DBG_CTRL is programmed on debug interface and debug_halted_o is actually asserted. During this window, ecall is executed, NPC is changed to PC(ecall) + 4 instead of the trap address.
- Program 'h1 to DBG_CTRL to enable single step mode, design executed instructions from PC(ecall) + 4.

Please check if it's a design issue.
Thanks
Tao

davideschiavone, 2018-03-09:
Dear Tao,

I tried to fix the issue here: d7f0f25bc4ab870196d8dc52e3b6b7f5bdad4a59

May you let me know if it solves the issue please?

Thanks a lot,
Best regards,
Davide

taoliug, 2018-03-09:
Just rerun the test and it's passing now, thanks for the fix

davideschiavone, 2018-03-09:
Thanks a lot, the fix has been pushed to both the master and pulpinov1 branch.

Best regards,
Davide

closed 2018-03-09 by davideschiavone