openhwfoundation/cv32e40p #32: The core jumped into the wrong address.

finalex, 2018-03-11:
When I change the riscv core from commit 6c928ad to f0180d6, something goes wrong.

This is the assembly code.
[image]

The incorrect waveform:
[image]
The correct waveform:
[image]

I trace the signal in riscv_controller.sv
`
    if ((jump_in_dec_i == BRANCH_JALR) &&
        (((regfile_we_wb_i == 1'b1) && (reg_d_wb_is_reg_a_i == 1'b1)) ||
         ((regfile_we_ex_i == 1'b1) && (reg_d_ex_is_reg_a_i == 1'b1)) ||
         ((regfile_alu_we_fw_i == 1'b1) && (reg_d_alu_is_reg_a_i == 1'b1))) )
    begin
      jr_stall_o      = 1'b1;
      deassert_we_o   = 1'b1;
    end
`
the pipeline does not stall when wait for the branch address.

davideschiavone, 2018-03-13:
Hello finalex,

thanks for reporting it.
Isn't the load_stall_o equal to 1 in the riscv_controller either?

Can you provide please the value of the signals that are checked in the if-statement in the controller for jr_stall_o and load_stall_o ?

Also, can you provide the instruction trace related to that part of program?

Thanks a lot, it will help me fixing it faster.
Best
Davide

finalex, 2018-03-14:
Hi,
The following is the waveform.
[image]

This is the trace log.
[trace_core_00_0.log](https://github.com/pulp-platform/riscv/files/1810628/trace_core_00_0.log)

Is there any document for helping understand the micro achitecture of core.

davideschiavone, 2018-03-14:
Hi, it looks really strange and it is similar to a previous issue here

https://github.com/pulp-platform/pulpino/issues/87

eventually it looked like a problem in testbench.

I noticed something strange in the waveform, I see the wb_ready_i = 1 and at the cycle after I expect the regfile_we_wb_i to be1 as well, that logic is handled in the ex_stage.

Can you provide waveforms about the ex_stage as well to see why this actually does not happen?

Thanks a lot
Davide

finalex, 2018-03-15:
Hi Davide,

The waveform is here.
[image]

finalex, 2018-03-16:
[image]
I check the logic about lsu_ready_ex_o and lsu_ready_wb. For single read， the two signal is the same as the data_gnt_i and data_rvalid_i. 
[image]
According to the logic above, I thought the jr_stall_o will be 0, between the data_gnt and data_rvalid. 
I'm confused with that...

finalex, 2018-03-16:
[image]
I found a change on the regfile_we_wb_o.

finalex, 2018-03-16:
I've tried to change it back, and it works for me.

davideschiavone, 2018-04-17:
Hi finalex, sorry for delay.

We (@stmach and I) are discussing it internally as this bug has been inserted trying to fix another one. We are trying to see if reverting this causes the other bug to return "unsolved". I suggest that you keep your fix in your branch for the moment until we solve the conflix!

Best and thanks
Davide

davideschiavone, 2018-04-19:
Dear finalex, my collegue @stmach double-checked that actually the fix he has done does not affect when reverting the issue you found! so we did it here: 3606a0247abb1a8b6161ad9e33a449e0632d338f

Thanks a lot for poiting it out,
Davide

closed 2018-04-19 by davideschiavone