openhwfoundation/cv32e40p #36: Responding to an instruction fetch in the same cycle leaves the core deadlocked

fabianschuiki, 2018-03-31:
I attached an instruction cache to the core that under certain circumstances can provide a responses to an instruction request in the same cycle (`instr_req`, `instr_gnt`, and `instr_r_valid` all in the same cycle). In this case the core ignores the response and deadlocks.

See the attached image: The core requests instruction word 0x1c000090, which is granted and responded to in the same cycle. However the core is now stuck, probably ignoring the response.
[image]

gautschimi, 2018-04-02:
> With an artificially delayed the response the core continues execution
> as expected.
> [image]
> I think this is a bug :smile:

This behaviour is correct. Rvalid has to come 1-x cycles after a grant

fabianschuiki, 2018-04-03:
Okay thanks, in that case I'll keep the circuitry that delays the response.

closed 2018-04-03 by fabianschuiki