openhwfoundation/cv32e40p #37: regfile_we_wb set from wb_ready_i caused load_stall to occur one instruction too late.

mcockrell-google, 2018-04-19:
In the controller the a pending load is indicated by WB not ready and regfile_we_wg high.  This occurs one cycle too late in successive loads. Keeping the prior logic of setting regfile_we_wb high when the regifle_we_lsu is high accurately indicates a pending load and wb in the correct cycle.

davideschiavone, 2018-04-25:
Hello Matt, thanks for reporting it.

Can you provide please an example and/or waveforms so that we can try to reproduce and fix the bug?

Thanks a lot
Davide

mcockrell-google, 2018-04-26:
Hi Davide - [image: image.png]

See the above screen shot of the trace log. There is a lbu instruction
executed (PC = b560). The branch is executed with the wrong value loaded in
x10. (see screenshot below)

As you can see the read stall for the load from 2392d occurs a cycle late.

[image: image.png]

[image: image.png]

Let me know if you need any more information.

Matt

davideschiavone, 2018-04-30:
Hi Matt, thanks a lot but I can't see the images, maybe it is easier if you send them to me by email.

Thanks a lot
Best
Davide

davideschiavone, 2018-06-21:
Solved here: 3606a0247abb1a8b6161ad9e33a449e0632d338f

closed 2018-06-21 by davideschiavone