openhwfoundation/cv32e40p #40: Nested Exception Support

ben-marshall, 2018-06-18:
Hello

I'm not sure if I am interpreting the manual [for this core](https://pulp-platform.org//wp-content/uploads/2017/11/ri5cy_user_manual.pdf) correctly, I hope you can help.

Section 12.3 talks about exception handling, specifically it says:

> RI5CY **does support nested interrupt/exception handling**. Exceptions inside interrupt/exception  
> handlers cause another exception, thus exceptions during the critical part of your exception handlers, i.e.  
> before having saved the MEPC and MESTATUS registers, will cause those register to be overwritten.

(emphasis mine)

To my understanding, to support nested exception handling (the RISC-V priv ISA spec calls it nested trap support: 3.1.7 of v1.10), one must include the `xPIE` and `xPP` fields of the `MSTATUS` CSR in order to properly recover the privilege mode and interrupt enabled states. It does not detail a hardware mechanism for saving `MSTATUS` and `MEPC`.

There is nothing to indicate hardware support for nested saving of the `MEPC` or the `MSTATUS` registers in the RI5CY user manual. This in my opinion is what explicit support for nested exception handling would be. Otherwise, as the manual says, the `MEPC` and `MSTATUS` registers are clobbered immediately by the first exception you get inside your handler, and unless you have already saved them via software, you can no longer get back to the code which caused the first exception.

So my question is this: does RI5CY support hardware assisted nested exception handling in machine mode, or does it only support nested exception handling the way any other core might, by requiring the `MEPC` and `MSTATUS` to be saved in time for any other exception which might occur?

Thanks, and let me know if anything doesn't make sense!

Ben

davideschiavone, 2018-06-28:
Hello Ben, 
indeed you are right, what Riscy supports is SW-assisted nested interrupt, i.e. the SW has to take care to save the MSTATUS and MEPC before any possible nested exceptions or interrupt.

I clarified this point writing in the document 
"RI5CY supports SW-assisted nested interrupt/exception handling"

New document submitted here: ba285006e0472ef88d3f5cf5f2b7b0312fb3f3e3
Thanks for pointing it out,
Cheers
Davide

closed 2018-06-28 by davideschiavone

ben-marshall, 2018-06-28:
Hi Davide
That's great, thanks for clarifying. :+1: