openhwfoundation/cv32e40p #47: Core performance counter is not accounting all cycles

haugoug, 2018-08-21:
When using the CYCLES event, not all cycles are accounted while all cycles should be accounted as soon as the core is clocked.
It seems to be only with special instructions like csr instructions.
The issue can be reproduced with this test: tests/pulp_tests/bugs/perf_cycle_lost
This happens on all chips even vega.

haugoug, 2018-08-21:
I also attach the instruction trace which shows the issue:
      11435920000000           69900 1c008ba4 780027f3 csrrs            x15, x0, 0x780      x15=00007d5b
      11435940000000           69901 1c008ba8 40f707b3 sub              x15, x14, x15       x15=0000001d x14:00007d78 x15:00007d5b
      11435960000000           69902 1c008bac fef04ce3 blt              x0, x15, -8         x15:0000001d
      11436019000000           69905 1c008ba4 780027f3 csrrs            x15, x0, 0x780      x15=00007d5f
      11436039000000           69906 1c008ba8 40f707b3 sub              x15, x14, x15       x15=00000019 x14:00007d78 x15:00007d5f
      11436058000000           69907 1c008bac fef04ce3 blt              x0, x15, -8         x15:00000019

As you can see the iteration takes 5 cycles but the core counter is increased by only 4 at each iteration

silabs-PaulZ, 2020-05-18:
The performance counters will be changed into HPM, as defined by the privileged specification. This bug was also still seen in this conversion.

The bug came about from a CSR read (CSRR) being assembled into a CSR read & set instruction where the data is a value of 0 (CSRRS w/rs=0 or CSRRSI w/zimm=0). This would, in turn, cause a write-back operation, essentially preventing an increment of the counter when reading. 

The fix has been implemented in the pull request #339.

Silabs-ArjanB, 2020-06-02:
Hi @haugoug Could you please close this ticket if you agree it has been fixed?

closed 2020-06-03 by haugoug