openhwfoundation/cv32e40p #49: Documentation???

luudee, 2018-09-23:
Hello,

is there a place where the parameters to the core are and the apu_master interface are explained?

Also, how do I configure the implementation? For example, I would like to implement it without a HW FPU.

Many Thanks!

rudi

Silabs-ArjanB, 2020-07-14:
Hi @luudee The top level parameters are explained here: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html#parameters. The APU interface will be (but has not yet been) explained here: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/apu.html. We have a separate task to create the APU doc here: https://github.com/openhwgroup/core-v-docs/issues/48. 

Could you please close this issue if you agree that it is partially answered and partially overlapping the mentioned core-v-docs issue? If not agreed, please let us know what type of info is still missing (and not already requested in core-v-docs/issues/48)

Silabs-ArjanB, 2020-07-23:
Hi @luudee We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-23 by Silabs-ArjanB