openhwfoundation/cv32e40p #50: Address is changed in ongoing request on instruction bus

felixmiller, 2018-11-05:
This is related to this pull request: https://github.com/pulp-platform/zero-riscy/pull/7
However the issues shows for both the zeroriscy and the ri5cy core.

When a branch instruction is encountered (and the branch is taken) the prefetch unit immediately puts the new address on the instruction bus when the prefetch unit is currently in the `WAIT_GNT` state.

Relevant sources:
ri5scy: https://github.com/pulp-platform/riscv/blob/master/rtl/riscv_prefetch_buffer.sv#L264
zeroriscy: https://github.com/pulp-platform/zero-riscy/blob/master/zeroriscy_prefetch_buffer.sv#L149

We see this issue causing problems in our implementation only when both the `branch_i` and the `instr_gnt_i` signal are asserted in the same cycle. However, also if the address is changed earlier (when the `instr_req_o` signal already has been asserted but before `instr_gnt_i` is received) the question remains if the protocol specification (from the user's manuals for the zeroriscy and the ri5cy core) allows this behavior.

[image]

The timing below shows this behavior when using the zeroriscy core. The address can change as late as the `instr_gnt_i` signal is received or at an earlier time in the `WAIT_GNT` state of the prefetch unit (2nd diagram).
[image]

[image]

wallento, 2018-11-05:
In my understanding, the address shouldn't change while `req` is asserted, correct?

gautschimi, 2018-11-05:
> This is related to this pull request: pulp-platform/zero-riscy#7
> timing2
> untitled
> untitled2

the address can change as long as the request is not granted. as soon as 
the request is granted (gnt_i received) the core expects data to the 
address which was valid when the gnt was received.

if the other side of your system is already serving a request to the 
previous address you should grant the request immediately. the data you 
can serve any number of cycles later together with the rvalid.

felixmiller, 2018-11-05:
Thanks for your quick response!

If this is intended behavior of the instruction bus, might this indicate an issue in the `core2axi` module then? According to the AXI specification it is not allowed to change the address once the `VALID` signal has been asserted. The `core2axi` module however directly maps the input address `data_addr_i` on the address signal (`ar_address_o`) of the read address channel of the AXI bus [(link to source](https://github.com/pulp-platform/core2axi/blob/master/rtl/core2axi.sv#L290)).

When the `core2axi` module is in `IDLE` [state](https://github.com/pulp-platform/core2axi/blob/master/rtl/core2axi.sv#L124) and a read access is [requested](https://github.com/pulp-platform/core2axi/blob/master/rtl/core2axi.sv#L127) via the `data_req_i` signal (`data_we_i` being [low](https://github.com/pulp-platform/core2axi/blob/master/rtl/core2axi.sv#L150)) a valid address is signaled to the AXI read channel via `ar_valid_o` ([source](https://github.com/pulp-platform/core2axi/blob/master/rtl/core2axi.sv#L151)). However, due to the [direct mapping](https://github.com/pulp-platform/core2axi/blob/master/rtl/core2axi.sv#L290), the address that is signaled to the AXI read channel still can change after the assertion of the valid signal.

The diagram below shows (just after the yellow cursor) how the address change on the instruction bus is forwarded to the AXI channel:

[image]

gautschimi, 2018-11-06:
> untitled3

Yes that might be correct. But I think core2axi is only used for the 
data interface, and the data interface holds the address constant and 
hence does not violate this constraint.

wallento, 2018-11-06:
So, there are two fixes:

1. We don't start an axi request unless the grant is asserted and the address is registered. From my limited understanding that should be understood by the core then, but we need to confirm that it doesn't lead to the same issue. It will also need a did of the spec then to highlight that the address can change.

2. A similar fix as the  for zero-riscy from above. That would essentially be a change of spec that address is stable for instructions too.

felixmiller, 2018-11-07:
@gautschimi yes, I guess you are right, it looks like you are only using the core2axi module for the LSU interface in your systems. We were a bit mislead by the statement in the documentation of the zero-riscy and ri5cy cores that the protocol was the same for both the instruction and data interface (Section 2.1).

Two add another option to @wallento 's list: Option 3 would be that you don't see this as an issue, since you have no plans on using the core2axi module for the instruction interface. May I suggest though, that you add a note in the documentation (.doc files), that the address might change again after asserting the `instr_req_o` signal for the instruction interface? Or maybe another timing diagram with the address signal being grey.

Regarding the two possible fixes mentioned by @wallento: 

For option 1, we have implemented a change ([this commit](https://github.com/felixmiller/core2axi/commit/69c3f28cbdc70889c006d05896f4c66c2bda944a)) in the `core2axi` module that immediatly grants the request on the instruction/data interface and keeps the signals on the AXI channels stable until the respective `x_ready` signal is asserted. We haven't thoroughly tested this yet but can create a PR once this is done, in case you are interested. This would still require a change/note in the documentation, the behavior of the cores is unchanged.

For option 2, we believe that this is fixed by [this](https://github.com/pulp-platform/zero-riscy/pull/7) PR (but only for the zero-riscy). If you plan to integrate this, this would also require the note in the documentation

davideschiavone, 2018-11-08:
Hello guys,

this is not a bug indeed. As @gautschimi said, the core has the right to change the address as long as it did not receive any grant.

Therefore, if you want to use the core2axi in the instruction bus, you need to change it as for example said by @wallento point 1.

@felixmiller  yes, I can add a note on the documentation.

So I don't think we need to fix Riscy as the GRANT is always associated with the current ADDRESS.
It is combinatorial.

I probably misunderstood the problem encountered with zero-riscy but now it is clear. 

For the core2axi, I put in discussion @fabianschuiki as he is the responsible of this IP :)
he would be happy for contributions I guess :+1:

wallento, 2018-11-08:
Yes, I think the fix for core2axi is the best way to move forward, I think we can close this issue and the PR on zeroriscy. We will iterate the proposed fix asap and then create a PR towards core2axi to have compliance with the AXI spec in a configurable manner (so that extra registers are only added on the instruction path).

Thanks for your help, guys!

Silabs-ArjanB, 2020-06-15:
Hi @felixmiller This issue has been solved now (see above pull request). If you agree could you please close this issue?

felixmiller, 2020-06-22:
Closing as resolved with referenced PR

closed 2020-06-22 by felixmiller