openhwfoundation/cv32e40p #51: bug for fclass instr ????

IC-Dream, 2018-11-20:
Hi,

The definition of  NAN from IEEE 754-2008 standard:
The value NaN (Not a Number) is used to represent a value that does not represent a real number. NaNs are represented by a bit pattern with an exponent of all ones and a non-zero mantissa. The sign bit can be 0 or 1 – it has no bearing.

There are two categories of NaNs:
QNaN (Quiet NaN) – arising when the result of an arithmetic operation is mathematically undefined. The MSB of the mantissa is '1' for this type of NaN.
SNaN (Signaling NaN) – used to signal an exception when an invalid operation is performed. The MSB of the mantissa is '0' for this type of NaN.

But the implement of fpu_core may exist some problem in riscv_alu.sv:
[image]

it seems that the RTL violates the definition of NAN. 

Thanks
Dream

FrancescoConti, 2018-11-20:
I think you're correct - however I transferred the issue to the riscy core repository, which is more appropriate.

IC-Dream, 2018-11-20:
@FrancescoConti Thanks a lot

stmach, 2019-05-07:
See #52

closed 2019-05-07 by stmach