openhwfoundation/cv32e40p #52: Bug for fle/flt/feq ???

IC-Dream, 2018-11-21:
Hi,

From the riscv_spec_v2.2 Page 52:
[image]

I created a directed case that set the operand to NAN like the below, but i didn't see that the NV flag was set  in RTL.
[image]

Thanks
Dream

IC-Dream, 2018-11-21:
The fmin.s/fmax.s also exist the same issue. thanks.

stmach, 2019-05-07:
Hello, I was just recently assigned.

Given the age of this issue, you're referencing a bug in the old FPU (https://github.com/pulp-platform/fpu). Since we're moving to a new FPU (https://github.com/pulp-platform/fpnew) which is used in the current and future releases of our cores, we are no longer updating the old FPU.

closed 2019-05-07 by stmach