openhwfoundation/cv32e40p #55: A problem with riscv_decoder.sv

hyf6661669, 2018-12-05:
In "riscv_decoder.sv", there are codes like:
```
 // special p.elw (event load)
        if (instr_rdata_i[14:12] == 3'b110)
          data_load_event_o = 1'b1;
```
When  ```instr[6:0] = OPCODE_LOAD``` or ```instr[6:0] = OPCODE_LOAD_POST```, the above code will be executed. But I find that only the instruction "LWU" in RV64I could lead to 
```
instr_rdata_i[14:12] = 3'b110
```
In "RV32IMFCXpulp", instr[14:12] could never be 3'b110.
Is this a little bug? Can you please explain the meaning of ```data_load_event_o```?

Thank you!

hyf6661669, 2018-12-20:
Can anyone have a look at my issue :( ??

davideschiavone, 2018-12-20:
Hello,

this instruction is meant to be used only in the cluster.

It is a custom extension that is used to load values from the cluster event unit of PULP.
This will clock gate the core, thus special behaviour is needed to differentiate this special load from the normal one.

Best
Davide

hyf6661669, 2018-12-21:
@davideschiavone  If I only use one RI5CY core for my design can I just ignore above codes?