openhwfoundation/cv32e40p #56: RISCY hangs when two continus mul was followed by ecall

IC-Dream, 2018-12-11:
Hi,

Here is a simple assembly case like the below:
[image]

the RISCY will hangs and can't finish, I noticed that the ctrl_fsm_cs of riscv_controller stop at FLUSH_EX state,  and can't  enter FLUSH_WB state. since the ex_valid_i would not go high and the pipeline was empty,  Also, the ecall would stall to fetch instr, so it is deadlock.

[image]

Thanks
Dream

IC-Dream, 2018-12-11:
when the ecall is at id_stage, the mul is just at ex_stage, then the ex_valid_i will pull down.

davideschiavone, 2018-12-19:
Thanks a lot for reporting it,

I will try to reproduce it and fix it!

Best
Davide

IC-Dream, 2018-12-20:
ebreak exists the same issue. Thanks.

hyf6661669, 2018-12-20:
@IC-Dream  @davideschiavone  I think the problem comes from "riscv_id_stage.sv":
```
else if(ex_ready_i) 
	  begin	
        // EX stage is ready but we don't have a new instruction for it,
        // so we set all write enables to 0, but unstall the pipe

        regfile_we_ex_o             <= 1'b0;
        regfile_alu_we_ex_o         <= 1'b0;
        csr_op_ex_o                 <= CSR_OP_NONE;
        data_req_ex_o               <= 1'b0;
        data_load_event_ex_o        <= 1'b0;
        data_misaligned_ex_o        <= 1'b0;
        branch_in_ex_o              <= 1'b0;
        apu_en_ex_o                 <= 1'b0;
        alu_operator_ex_o           <= ALU_SLTU;
        mult_en_ex_o                <= 1'b0;               //when
      end 
```

In "riscv_controller", when ```mret_insn_i | uret_insn_i | ecall_insn_i | pipe_flush_i | ebrk_insn_i | illegal_insn_i = 1```, ```halt_id_o``` will become ```1``` so ```riscv_id_stage.id_valid_o = 0```. But in this period, the "mul" instruction can be completed so we get ```riscv_id_stage.ex_ready_i = 1``` and ```riscv_id_stage.mult_en_ex_o <= 0```. As a result, in next period, we get ```riscv_ex_stage.mult_en_i = riscv_ex_stage.ex_valid_o = riscv_controller.ex_valid_i = 0```, so ```riscv_controller.ctrl_fsm_cs``` just stucks in ```FLUSH_EX```.

IC-Dream, 2018-12-20:
Yea, I agree with you, and i do the following hack in my local. please help to check.
Related files: 1. riscv_id_stage.sv 2. riscv_ex_stage.sv 3. riscv_core.sv
when a ecall enter the id_stage, i think the alu_en_ex_o also shoule be disable like the mul_en_ex_o, and add one siginal:system_insns_dec in id_stage and just pipeline to ex_stage.
[image]

[image]

Then, the ex_valid_o should also think about the system_insns_ex.
[image]

The core would not hangs anymore, but i am not sure if introduce any side effect.

Thanks
Dream

hyf6661669, 2018-12-20:
Actually similar issue may also happen because in "riscv_ex_stage":
```
assign ex_valid_o = (apu_valid | alu_en_i | mult_en_i | csr_access_i | lsu_en_i) & (alu_ready & mult_ready & lsu_ready_ex_i & wb_ready_i);
```
And ```Load/Store``` instructions may also lead to similar problems if they are followed by ```ECALL/EBREAK ...```.

If we use ```riscv_ex_stage.ex_ready_o``` instead of ```riscv_ex_stage.ex_valid_o``` for ```riscv_controller```, I think this bug could be fixed.

IC-Dream, 2018-12-20:
Yea, I also doubt the load/store would exist same issue, but i did some random test, it seems that it work well.

hyf6661669, 2018-12-20:
At present I can't understand the code related with ```APU```, so I don't know whether instructions that want to use ```APU``` can lead to this issue. Because ```riscv_ex_stage.ex_valid_o``` is controlled by ```riscv_ex_stage.apu_valid```, which seems risky.

davideschiavone, 2018-12-20:
Dear all,

I confirm that is a bug.

I tried to solve it in this branch
7f10f5cd3d2184bc1932528b8ff955d2de4856f9
 but I will merge it to the master after all our internal tests pass to be sure I haven't broken anything else due to the sensitivity of this pipeline control signal.

Best
Davide

davideschiavone, 2019-01-07:
Can you confirm whether this fix works or the test is still failing?

Thanks
Davide

davideschiavone, 2019-01-09:
No problem :+1: nice to hear that :) I am about to merge it to the master

7e8495f8fc5a8f3b308d40df470f690717b5dc9e

closed 2019-01-09 by davideschiavone