openhwfoundation/cv32e40p #60: CSR access via debug interface

ajaykrishna1111, 2019-01-05:
Hi,
I tried to enable the performance counters via the debug interface; but the value is not latched in the PCCR_q as the dbg_gnt_o holds only for half cycle for CSR access. 
[image]

silabs-PaulZ, 2020-05-28:
Hi Ajay,

There have been some changes since your last post. The debug interface to the core only comprises of debug_req_i. There was a bug that a single clock pulse on debug_req_i would not properly enter debug mode. This has been fixed and is awaiting a merge (#334 ).

Also, the legacy performance counters have been deprecated as we are now using the Hardware Performance Counters as specified in the RISCV privileged specification. (#339 )

With these updates, you can see that we can enter debug_mode with a single pulse request on debug_req_i and can access these new HPM CSR while in debug mode.
[image]

Note: At this time, we are not planning to implement the stopcount bit on CV32E40P. It is tied to a default value of 0, implying performance counting occurs during debug and non debug mode. 

We are waiting for the merge of those pull request. When those are merged in, will this address the concern or issue you are reporting.

Silabs-ArjanB, 2020-07-14:
Hi @ajaykrishna1111 , the mentioned pull requests have been merged. Could you please close this issue if you agree it has been addressed satisfactorily or let us know of further questions?

Silabs-ArjanB, 2020-07-23:
Hi @ajaykrishna1111 We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-23 by Silabs-ArjanB