openhwfoundation/cv32e40p #62: Adding new instructions

hyf6661669, 2019-03-07:
Hi, I want to add some new custom instructions to RI5CY. I have spent lots of time on goole but couldn't find a satisfying answer about how to add new instructions to the riscv-gnu-toolchain. Can you give some hints?
Much thanks!

osta1, 2020-09-03:
Hello,

I am also facing the same issue; I want to add some new instructions to pulp-riscv-gnu-toolchain, but I am not able to find some suitable answers.
Thank you for informing me, in case you have solved the problem.

Thanks.

MikeOpenHWGroup, 2020-09-03:
Hi @hyf6661669, and @osta1.  Thanks for your query.   I _believe_ you are asking about the toolchain (that is, compiler/assember/linker) and not the core itself.  If that is the case, this issue should be posted to [core-v-sw](https://github.com/openhwgroup/core-v-sw).

If you agree, please close this issue and create a new one there.

closed 2020-09-03 by hyf6661669