openhwfoundation/cv32e40p #63: Illegal compressed instruction (9042) causes a parallel case fail on the unique case decoding block

ntuszynski, 2019-03-18:
An incoming illegal compressed instruction (value=9042) sets the ebrk_insn_i and illegal_insn_i flags simultaneously. The flags are used in an unique case block where the illegal_instruction branch is not executed, instead the ebreak_instruction branch is taken.
[image]

davideschiavone, 2019-03-18:
Hello, you are right.

I am fixing this

Best
Davide

davideschiavone, 2019-03-18:
can you try again with this commit please?

2bad1016918b7195aebf4ed71f7f94ad3e0ce9d2

thanks for catching and reporting it
Davide

ntuszynski, 2019-03-18:
My pleasure! Will try it for sure.
Thanks for quick reply.
BR,
Nicolae

davideschiavone, 2019-04-12:
Any news @ntuszynski ?

ntuszynski, 2019-04-24:
This one is fixed!

closed 2019-05-06 by davideschiavone