openhwfoundation/cv32e40p #64: What does "APU" mean?

hyf6661669, 2019-03-22:
It seems that the signals related to "APU" are also related to "FPU". Some input and output signals of fpu_private in riscv_ex_stage have the prefix with "apu_":
```
fpu_private fpu_i
             (
              .clk_i          ( clk               ),
              .rst_ni         ( rst_n             ),
              // enable
              .fpu_en_i       ( apu_req           ),
              // inputs
              .operand_a_i    ( apu_operands_i[0] ),
              .operand_b_i    ( apu_operands_i[1] ),
              .operand_c_i    ( apu_operands_i[2] ),
              .rm_i           ( apu_flags_i[2:0]  ),
              .fpu_op_i       ( fpu_op_i          ),
              .prec_i         ( fpu_prec_i        ),
              // outputs
              .result_o       ( apu_result        ),
              .valid_o        ( apu_valid         ),
              .flags_o        ( fpu_fflags_o      ),
              .divsqrt_busy_o ( fpu_busy          )
              );
```
Can you tell me what does "APU" mean?

fabianschuiki, 2019-03-25:
We use APU to refer to a bunch of Auxiliary Processing Units. The APU mechanism in the core provides a means to fork instructions out of the pipeline and push them into a separate unit. The FPU attaches to this mechanism, for example. Multiple processors may also share seldomly-used resources via this port.

hyf6661669, 2019-03-25:
> We use APU to refer to a bunch of Auxiliary Processing Units. The APU mechanism in the core provides a means to fork instructions out of the pipeline and push them into a separate unit. The FPU [...]

Clear explanation, thanks!

closed 2019-03-25 by hyf6661669