openhwfoundation/cv32e40p #68: State of the FPU

jm4rtin, 2019-04-24:
There was an older FPU block that was being used ([pulp-platform/fpu](https://github.com/pulp-platform/fpu)) and a new block that is "unreleased" and has "no support" ([pulp-platform/fpnew](https://github.com/pulp-platform/fpnew)). I have a few questions about the status of the FPU:

1. How close to being released is the new FPU?
2. How well verified is the new FPU?
3. How difficult would it be to instantiate the old FPU in the design on the master branch?

stmach, 2019-05-07:
Hello,

> How close to being released is the new FPU?

We are in the process of cleaning up and adding documentation to the new FPU, hence the "in development" notice is still up. We expect that to be done in a couple of weeks. The design itself won't change much until then, and no major features are being added for now.

> How well verified is the new FPU?

The FPU was tested against the RISC-V tests as well as RISC-V torture suite, as well as isolated tests of different modules against a software golden model (using native x86 FP instructions). The FPU is fully IEEE compliant to our knowledge, but the iterative divider/square root unit (which was also used in the old FPU) has some rounding issues that will have to be resolved with a later release.

> How difficult would it be to instantiate the old FPU in the design on the master branch?

Changes would be needed mainly to the decoder and ex_stage modules of the core to instantiate the old FPU and decode for the control inputs it needs. It should be doable with moderate effort by cross-referencing (or straight up reverting to) older versions of these modules from past releases.
However, please note that the old FPU potentially has some arithmetic bugs and we're no longer supporting or updating the old FPU.

davideschiavone, 2020-05-17:
@jm4rtin , this issue seems resolved. Can we close it?
thx,
Davide

jm4rtin, 2020-06-09:
Yep.

closed 2020-06-09 by jm4rtin