openhwfoundation/cv32e40p #69: Documentation of DBG_CAUSE

stefanct, 2019-05-03:
The documentation of the the `cause` part of the `DBG_CAUSE` register is lacking a lot of important details. For example it is completely unclear how the exceptions are encoded. One might think that they are encoded like the interrupt enables in its neighbor register `DBG_IE` - and this apparently was the case when the [debug bridge](https://github.com/pulp-platform/debug_bridge/blob/518c46320bdca8ab8d5a3fdcd14b1231afd02c22/rsp.cpp#L786) was written. AFAICT it gets it completely wrong (I am trying to fix that ATM).

However, the values have nothing to do with `DBG_IE` at all! They are coming from [riscv_defines.sv](https://github.com/pulp-platform/riscv/blob/08a95b12db0912e5ca3bb28638b5cd18182ab45f/rtl/include/riscv_defines.sv#L329) via `exc_cause_o` in the `FLUSH_WB` state in [riscv_controller.sv](https://github.com/pulp-platform/riscv/blob/08a95b12db0912e5ca3bb28638b5cd18182ab45f/rtl/riscv_controller.sv#L705). They are coming from an ancient RISCV debug spec draft that had this cause register (e.g. v0.4) but was replaced later with another mechanism to determine the halt reason.

I am working with Pulpino and I have noticed that some things in the `FLUSH_WB` state have changed in the upstream repo but I don't think it matters. I would have submitted a change to the docs themselves... but I won't touch `.doc` files, sorry.

silabs-PaulZ, 2020-05-26:
The DCSR.cause field seems to be correctly reporting the causes as defined in the riscv debug specification version 0.13.2. In particular, the section 4.8.1 has the following:
[image]

The implementation seems to be correctly reporting the correct cause, as can be seen here:
https://github.com/openhwgroup/cv32e40p/blob/d830c29a2ce7ca43e876c4488f22755630d8f126/rtl/riscv_controller.sv#L932-L937
and
https://github.com/openhwgroup/cv32e40p/blob/d830c29a2ce7ca43e876c4488f22755630d8f126/rtl/riscv_controller.sv#L951-L958
If you agree, can you please close this ticket?

stefanct, 2020-05-26:
Your comments are rather independent from my bug report since your are talking about the code and the debug specification while I was talking about the old ri5cy documentation in `doc/user_manual.doc`. However, since the latter was removed in df2452e1 this can probably indeed be closed. I have not checked the new documentation at https://github.com/openhwgroup/core-v-docs/tree/master/cores/cv32e40p/user_manual but I presume too much has changed that this could possibly be of any relevance anymore.

closed 2020-05-26 by stefanct