openhwfoundation/cv32e40p #71: Load/Store exceptions completely broken?

stefanct, 2019-05-15:
If I am not completely mistaken the exceptions of loads and stores are completely defunct since the "new" exception controller was added (around 3d957e5).

If I decypher how the respective bits (`LAF` (5) and `SAF` (7)) of the `DBG_IE` register are handled starting from `riscv_debug_unit.sv` (where they are stored `settings_q` and forwarded in `settings_o` to the controller in `riscv_controller.sv`) I end up nowhere. Other settings influence how the controller sets the `dbg_trap_o` signal but the `DBG_SETS_ELSU` setting (which combines `LAF` and `SAF`) is not used anyway at all.

In earlier versions that bit was taken into account in `exc_controller.sv` where it was used to determine the simple `trap_o` signal in conjunction with `lsu_store_err_i` and `lsu_load_err_i`. The latter signals came from the LSU and were actually triggering the interrupt by setting the `req_int` signal in the previous exception controller. Interestingly these `lsu_store_err_i` and `lsu_load_err_i` signals survived the port to the new controller and are still set in the LSU and forwarded to the ID stage but not passed further down in ID to the instance of `riscv_controller` and thus completely ignored(!)

Since I need this kind of exception handled for a demo in two weeks I am currently a little bit freaked out :(

stefanct, 2019-05-15:
Pinging @davideschiavone with this comment since he made the majority of the respective changes and probably knows best.

bluewww, 2019-05-16:
Which version of RI5CY are you referring to? The master branch has no `riscv_debug_unit.sv`.

stefanct, 2019-05-16:
Yes, sorry I was under shock last night ;) I am working on PULPino thus I was referring to `pulpinov1` aka 18c10e3.

bluewww, 2019-05-16:
I'm not too familiar with this version. I'll try to get Davide on this.

stefanct, 2019-06-29:
It has been a while and the workshop is over (I couldn't be there unfortunately). Have you looked at this, @davideschiavone? I am also not entirely sure what the governance change of ri5cy (that is now maintained by lowRISC if I understood correctly) has on issues in this tracker.

davideschiavone, 2019-07-01:
Hello @stefanct ,

so RI5CY will be soon join the CORE-V family and it will be maintained at industry-level standards by Open-HW (https://www.openhwgroup.org/)

LowRISC took over zero-riscy instead, which is now called Ibex: https://github.com/lowRISC/ibex

The debug unit is now completely changed as we moved to the RISC-V compliant spec and as you pointed out, the error coming from bus have been also removed. 
This part could be part again of RI5CY in the next future, but I can't give any specific timeline yet sorry.

We handled bus-errors by connecting the error signals to the interrupt controller of the platform, from there, an interrupt is raised to the core. The core has also a RISC-V compliant PMP which instead triggers exceptions synchronously in case one of the rules is violated.

I hope this helps you,
best
Davide

davideschiavone, 2020-05-17:
@stefanct can we close the issue now?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it has been answered at least 7 days ago. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB