openhwfoundation/cv32e40p #72: Verilator simulation broken

wallento, 2019-05-21:
Today I wanted to run the Verilator simulation, but it seems broken. I think it relates to those yml files and the fpnew package not being found:

```
verilator -O3 -CFLAGS "-O3 -g3 -std=gnu++11" \
                  -Wno-CASEINCOMPLETE -Wno-LITENDIAN -Wno-UNOPT \
          -Wno-UNOPTFLAT -Wno-WIDTH -Wno-fatal --top-module top \
          --Mdir obj_dir --trace -DPULP_FPGA_EMUL -cc \
          +incdir+../rtl/include cluster_clock_gating.sv dp_ram.sv ram.sv top.sv ../rtl/include/apu_core_package.sv ../rtl/include/riscv_defines.sv ../rtl/include/riscv_tracer_defines.sv ../rtl/register_file_test_wrap.sv ../rtl/riscv_alu.sv ../rtl/riscv_alu_basic.sv ../rtl/riscv_alu_div.sv ../rtl/riscv_compressed_decoder.sv ../rtl/riscv_controller.sv ../rtl/riscv_cs_registers.sv ../rtl/riscv_decoder.sv ../rtl/riscv_int_controller.sv ../rtl/riscv_ex_stage.sv ../rtl/riscv_hwloop_controller.sv ../rtl/riscv_hwloop_regs.sv ../rtl/riscv_id_stage.sv ../rtl/riscv_if_stage.sv ../rtl/riscv_load_store_unit.sv ../rtl/riscv_mult.sv ../rtl/riscv_prefetch_buffer.sv ../rtl/riscv_prefetch_L0_buffer.sv ../rtl/riscv_register_file.sv ../rtl/riscv_core.sv ../rtl/riscv_apu_disp.sv ../rtl/riscv_L0_buffer.sv ../rtl/riscv_pmp.sv testbench.cpp --exe
%Error: ../rtl/include/riscv_defines.sv:391: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/include/riscv_defines.sv:391: Perhaps 'fpnew_pkg' is a package which needs to be predeclared? (IEEE 2017 26.3)
%Error: ../rtl/include/riscv_defines.sv:392: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/include/riscv_defines.sv:393: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/include/riscv_defines.sv:452: syntax error, unexpected endpackage
%Error: ../rtl/riscv_decoder.sv:238: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:241: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:242: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:243: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:679: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:684: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:689: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:694: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:703: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:714: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:726: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:733: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:741: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:749: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:759: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:773: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:782: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:796: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:815: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:832: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:835: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:837: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:842: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:847: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:852: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:859: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:864: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:869: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:874: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:889: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:897: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:905: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:914: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:923: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:933: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:942: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:952: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:961: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:977: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:980: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:983: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:989: syntax error, unexpected ::, expecting ';'
%Error: ../rtl/riscv_decoder.sv:994: syntax error, unexpected ::
%Error: ../rtl/riscv_decoder.sv:1001: syntax error, unexpected ::
%Error: ../rtl/riscv_decoder.sv:1007: syntax error, unexpected default
%Error: Exiting due to too many errors encountered; --error-limit=50
```

bluewww, 2019-05-23:
Thanks for the report. Actually there is also an issue when setting `PULP_SECURE` to 1 because verilator doesn't understand all of SystemVerilog.

altanary, 2019-05-26:
I think it means that the macro fpnew_pkg is missing to be found. 
furthermore, the file named fpnew is not in project.

bluewww, 2019-05-26:
https://github.com/pulp-platform/fpnew. I think of a way to pull the dependencies for the verilator testbench.

wallento, 2019-05-26:
Isn't that what the Bender file is for? I can also provide a PR with the fusesoc setup if needed. That's pretty simple..

bluewww, 2019-05-26:
Indeed your right about the dependency missing in the bender file. I suppose I was about to hack to together a bash script to setup the tb dependencies because bender (the program that manages our ip dependencies) is not released yet. 

Anyway I was not aware of fusesoc, PR is welcome I can review and merge it.

tcmichals, 2019-08-07:
I was also trying to generate verilator-model, used cargo bender, but, it didn't build.  Is there a way to just provide some documentation on how to pull other IP blocks together to build the verilator-model?

uge, 2019-08-14:
Pulling the testbench code and the following diff allow it to build

```
--- a/verilator-model/Makefile
+++ b/verilator-model/Makefile
@@ -42,6 +42,7 @@ VSRC = cluster_clock_gating.sv            \
        dp_ram.sv                          \
        ram.sv                             \
        top.sv                             \
+       ../tb/core/fpnew/src/fpnew_pkg.sv  \
        ../rtl/include/apu_core_package.sv     \
        ../rtl/include/riscv_defines.sv        \
        ../rtl/include/riscv_tracer_defines.sv \
```

PR #146 (2019-09-04): Fix various verilator related problems in the testbenches
* Move verilator-model to tb/ directory and adjust Makefile
* Add -Wno-BLKANDNBLK to disable warning
  This allow us to use the PULP_SECURE=1 code path
* Add missing fpnew dependency to verilator-model makefile

Fixes #72 
Related #141
closed 2019-09-04 by bluewww