openhwfoundation/cv32e40p #79: Error when running Makefile in tb/dm

zdogs, 2019-06-20:
Hi,

I get an error when running Makefile in tb/dm : 
/riscv-pulp/bin/riscv32-unknown-elf-gcc -march=rv32imc  -o prog/test.elf -w -Os \
        -T prog/link.ld  \
        prog/test.c prog/syscalls.c prog/start.S
/riscv-pulp/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/lib/crt0.o: In function `.L0 ':
(.text+0x10): undefined reference to `_fbss'
collect2: error: ld returned 1 exit status
make: *** [prog/test.elf] Error 1

davideschiavone, 2019-06-20:
Thanks, we will have a look into this as soon as possible :)

bluewww, 2019-08-03:
I can't reproduce this issue with the pulp toolchain. I have made some changes recently, maybe its fixed for you. Reopen if you still need assistance.

closed 2019-08-03 by bluewww

shivampotdar, 2021-02-01:
I am still seeing this issue with the dm testbench in CV32E40P (tag `pulpissimo-v3.4.0`). My PULP-GCC build is based on commit #`49dea19` from pulp-riscv-gnu-toolchain.
The Makefile runs successfully with SiFive prebuilt toolchains. But that's not a good idea for XPULP  I believe.