openhwfoundation/cv32e40p #80: WAIT_RVALID_EX_STALL dead state?

davideschiavone, 2019-06-20:
WAIT_RVALID_EX_STALL seems to be a DEAD state in the LSU

davideschiavone, 2019-12-11:
removed in new_pipeline branch. Under evaluation

3bfe1ea820e60e3239a3d9b3c8ef403c45d4c3c1

Silabs-ArjanB, 2020-06-15:
Hi @davideschiavone Can you close this issue now (has been solved with above PR)?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB