openhwfoundation/cv32e40p #81: No Exception when code in User mode attempts to write in Machine mode register

zdogs, 2019-06-20:
Hi,

When code in User mode tries to write in a Machine mode reserved register, the write is not performed (hopefully) but the code does not go in exception vector (it should be). 
Here is a piece of code that shows the issue:

        ##;----------------------------------------------------------------
        ##; Set user program address in uepc
        ##;----------------------------------------------------------------
        la      a0,user_prog
        csrw    uepc,a0
        ##;----------------------------------------------------------------
        ##; Set user handler address in uvtec
        ##;----------------------------------------------------------------
        la      a1,u_handler
        csrw    utvec,a1
        ##;----------------------------------------------------------------
        ##; Set machine handler address in mvtec
        ##;----------------------------------------------------------------
        la      a1,m_handler
        csrw    mtvec,a1
        ##;----------------------------------------------------------------
        ##; Switch to user mode
        ##;----------------------------------------------------------------
        uret

        
user_prog:
        ##; Global interrupt enabled
        csrr    a1,ustatus
        li      a2,USTATUS_UIE
        or      a1,a2,a1
        csrw    ustatus,a1
        
        ##;----------------------------------------------------------------
        ##; FORBIDDEN INSTRUCTION : GO TO EXCEPTION VECTOR (@MTVEC) => DOES NOT WORK
        ##;----------------------------------------------------------------
        li      a1,0xFFFFFFFF
        csrw    mepc,a1       ##; write not executed but the program does not go to exception
        nop
        nop
        nop
        j       laberror
        
after_forbidden_instr:

davideschiavone, 2019-06-24:
Thanks for reporting it,

please find here: 1449ba6a8be1a0c2b061517d270bd9229ec8540a

my fix. Can you try to update the core at that version and see whether it fix the bug?

It should be ok.

Waiting for you feedback before integrating the fix into the master branch.
Best regards,
Davide

zdogs, 2019-06-24:
Hi Davide,

It works fine. The core enters exception now. 

However when trying to execute mret in user mode (forbidden) we get a simulator warning: 

  ncsim: *W,MCONDE: Unique case violation:  Multiple matching case item expressions at {line=773:pos=25 and line=783:pos=22}.
            File: /riscv_controller.sv, line = 750, pos = 20
           Scope: top.riscv_core_i.id_stage_i.controller_i
            Time: 155481150 PS + 59

This is due to illegal_insn_i and mret_insn_i conditions are both true creating a conflict for ctrl_fsm_ns. 
I think this is not clean and should be fixed. 

Regards

davideschiavone, 2019-06-24:
You are right thanks! I fixed here: b24b1029750f333db6e8ffe6a681d87b71411cbe

this improves indeed the quality of the RTL code.
I also removed a "case" statement in the XRET_JUMP state in the controller that I added by mistake.

is this solving your warning?

zdogs, 2019-06-24:
Yes ! no warning any more.

Thanks

davideschiavone, 2019-06-24:
Thanks! it is now pushed to master! b24b1029750f333db6e8ffe6a681d87b71411cbe

closed 2019-06-24 by davideschiavone