openhwfoundation/cv32e40p #83: Decoder bug for casting instruction FP to INT

OttG, 2019-06-25:
Inside the decoder at casting instruction FP to INT (it can be found at the comment // fcvt.ifmt.fmt - FP to Int Conversion) the decoder is only setting fpu_fmt_o which is the destination format and instead it should be setting fpu_fmt2_o which is the source format. 
To fix it it should be checked the bits of instr_rdata_i[26:25] that contain the source format. Also for the 16bit extension the bit at instr_rdata_i[14] is needed to be checked to descriminate between 16 and 16alt.

stmach, 2019-06-26:
Ciao,
I was told you already have a code fix for this, is that right? In that case, please open a pull request and assign me. Otherwise let me know!

OttG, 2019-06-26:
I couldn't assign you since I don't have write permissions here, I had to fork it first and then do the pull request.

OttG, 2020-11-18:
Solved

closed 2020-11-18 by OttG