openhwfoundation/cv32e40p #89: tb/core/master/firmware/stats.c: can't read number of cycles, cycles always=0;

shjdzc12, 2019-07-17:
In function "void stats(void)", it includes this command:
"__asm__ volatile("csrr %0, 0x780" : "=r"(num_cycles));"
I think it aims to read the value of PCRR0 register(address 0x780), which counts the number of cycles the core was active. However, num_cycles is always 0 and I don't know the reason of this problem.

shjdzc12, 2019-08-05:
Finally, I change the rtl code riscv_cs_registers.sv. 
I assign pcer=1, and then the "register 0x780" can read the cycles of the core.
It seems that the pcer register can't be enabled automatically.

davideschiavone, 2019-08-05:
The PCER should be enabled in SW before starting the measurments and not by changing the RTL.

Is it done in the firmware?

shjdzc12, 2019-08-05:
> The PCER should be enabled in SW before starting the measurments and not by changing the RTL.
> 
> Is it done in the firmware?

I didn't find it in the firmware. And when I add"csrw 0x7a0,1" by myself, it still doesn't work.

shjdzc12, 2019-08-05:
> The PCER should be enabled in SW before starting the measurments and not by changing the RTL.
> 
> Is it done in the firmware?

void stats(void)
{
    unsigned int num_cycles, num_instr;
    /* ideally we could use this */
    // __asm__ volatile ("rdcycle %0; rdinstret %1;" : "=r"(num_cycles),
    // "=r"(num_instr));
    /* riscy specific */
    __asm__ volatile("csrr %0, 0x780" : "=r"(num_cycles));
    __asm__ volatile("csrr %0, 0x781" : "=r"(num_instr));
    print_str("Cycle counter ........");
    stats_print_dec(num_cycles, 8, false);
    print_str("\nInstruction counter ..");
    stats_print_dec(num_instr, 8, false);
    print_str("\nCPI: ");
    stats_print_dec((num_cycles / num_instr), 0, false);
    print_str(".");
    stats_print_dec(((100 * num_cycles) / num_instr) % 100, 2, true);
    print_str("\n");
}

Above code is in the tb/core/firmware/status.c. However, 0x780 doesn't calculate the cycles and 0x781 doesn't record the number of instructions.

davideschiavone, 2019-08-05:
I think this function is to read the stat. Before measuring (before running the benchmark) you need to enable the PCER

bluewww, 2019-08-05:
Yeah I don't think I enable the PCER in the firmware. Let me have a look at it.

bluewww, 2019-08-05:
Fixed in #114

closed 2019-08-05 by bluewww