openhwfoundation/cv32e40p #90: References for Compliance tests missing

noytzach, 2019-07-17:
I managed to compile and run the testbench supplied under tb/core.

According to [tb/core/README.md](https://github.com/pulp-platform/riscv/tree/master/tb/core#ri5cy-core-testbench) it runs riscv-tests as well as riscv-compliance-tests, however I couldn't find the *.reference_output files as in the original [riscv-compliance-tests](https://github.com/riscv/riscv-compliance).
Inspecting the Makefile, I couldn't find where the test signature is compared to an expected value as desribed [here](https://github.com/riscv/riscv-compliance/tree/master/doc#25-the-test-signature).

Can you help me understand how the compliance test results are verified?

Thank you
@noytzach

bluewww, 2019-07-17:
Indeed we don't do this. This would be a welcome improvement. 

On the other hand most targets in `riscv-compliance/riscv-target` don't do anything at all:
https://github.com/riscv/riscv-compliance/blob/master/riscv-target/spike/compliance_io.h
https://github.com/riscv/riscv-compliance/blob/master/riscv-target/rocket/compliance_io.h
etc.
but we check if the assertions fail or not (and print a corresponding message)

bluewww, 2019-08-29:
Patch incoming
https://github.com/riscv/riscv-compliance/pull/64/files

PR #138 (2019-08-30): Tb enhancements
Fixes #90 
* Allow user to provide maximum simulation time in cycles 
* Add signature dumping for riscv-compliance
Compare with https://github.com/riscv/riscv-compliance/pull/64
commit b80e16c7 (2019-08-30): Add signature dumping for riscv-compliance

Resolves #90

Signatures will be printed to stdout and can be additionaly written to a
file by providing a path like this: `+signature=your_path`.
closed 2019-08-30 by bluewww