openhwfoundation/cv32e40p #91: compile error when compile tb/core/custom/hello_world.c

shjdzc12, 2019-07-19:
In tb/core/Makefile, following command is included:

/home/zhangchao/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imc -o custom/hello_world.elf -w -Os -g -nostdlib	-T custom/link.ld -static	custom/crt0.S	custom/hello_world.c custom/syscalls.c custom/vectors.S  -I /home/zhangchao/riscv/riscv32-unknown-elf/include -L /home/zhangchao/riscv/riscv32-unknown-elf/lib -lc -lm 

According to README.md, this command is to compile the hello_world.c and get hello_world.elf. However, When I run this command, following errors will be reported:
 
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text+0x10): undefined reference to `sbrk'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-writer.o): In function `_write_r':
writer.c:(.text+0x14): undefined reference to `write'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-closer.o): In function `_close_r':
closer.c:(.text+0x10): undefined reference to `close'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-lseekr.o): In function `_lseek_r':
lseekr.c:(.text+0x14): undefined reference to `lseek'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-readr.o): In function `_read_r':
readr.c:(.text+0x14): undefined reference to `read'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-fstatr.o): In function `_fstat_r':
fstatr.c:(.text+0x12): undefined reference to `fstat'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-isattyr.o): In function `_isatty_r':
isattyr.c:(.text+0x10): undefined reference to `isatty'
collect2: error: ld returned 1 exit status
Makefile:213: recipe for target 'custom/hello_world.elf' failed
make: *** [custom/hello_world.elf] Error 1

I install my toolchain according to  "pulp-platform/pulp-riscv-gnu-toolchain"( Installation (Pulp) section). I want to know why I get these errors and how to solve this problem.

shjdzc12, 2019-07-19:
"/home/zhangchao/riscv" is my installation path for toolchain. I'm sorry if this causes confusion.

bluewww, 2019-07-19:
You are not linking against newlib. Install and use riscv-gnu-toolchain for an easy way to use gcc+newlib. Ill have a look at the pulp-sdk newlib path.

shjdzc12, 2019-07-19:
> You are not linking against newlib. Install and use riscv-gnu-toolchain for an easy way to use gcc+newlib. Ill have a look at the pulp-sdk newlib path.

Thank you for your reply. But "the newlib section in riscv/riscv-gnu-toolchain" just guide me to install the 64-bit toolchain. I want to know how to install the 32-bit newlib toolchain because ri5cy needs the one with 32-bit.

bluewww, 2019-07-19:
`./configure --prefix=$YOUR_INSTALLDIR --with-arch=rv32gc --with-abi=ilp32`

shjdzc12, 2019-07-19:
> use something like:
> `./configure --prefix=$YOUR_INSTALLDIR --with-arch=rv32gc --with-abi=ilp32`

Thank you for your solution

shjdzc12, 2019-07-20:
> use something like:
> `./configure --prefix=$YOUR_INSTALLDIR --with-arch=rv32gc --with-abi=ilp32`
Sorry for my interruption. I use the followingsuch commands to install the 32-bit toolchain:

$./configure --prefix=$YOUR_INSTALLDIR --with-arch=rv32imc --with-abi=ilp32
$make

You set the option "arch" as "rv32gc", but ri5cy supports rv32imc, so I modify rv32gc to rv32imc.
However, when I compile the hello_world.c, previous problems still exist. tell me serveral undefined references:

/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-sbrkr.o): In function _sbrk_r': sbrkr.c:(.text+0x10): undefined reference to sbrk'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-writer.o): In function _write_r': writer.c:(.text+0x14): undefined reference to write'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-closer.o): In function _close_r': closer.c:(.text+0x10): undefined reference to close'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-lseekr.o): In function _lseek_r': lseekr.c:(.text+0x14): undefined reference to lseek'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-readr.o): In function _read_r': readr.c:(.text+0x14): undefined reference to read'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-fstatr.o): In function _fstat_r': fstatr.c:(.text+0x12): undefined reference to fstat'
/home/zhangchao/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-isattyr.o): In function _isatty_r': isattyr.c:(.text+0x10): undefined reference to isatty'

Are there still some problems with my toolchain installation?

bluewww, 2019-08-03:
Duplicate of #92, closing.

closed 2019-08-03 by bluewww