openhwfoundation/cv32e40p #92: unable to generate hello_world.elf

shjdzc12, 2019-07-22:
I download necessary file  from "pulp-platform/pulp-riscv-gnu-toolchain" and install the toolchain according to the "newlib section". Here is my command to install the toolchain:

$./configure --prefix=/home/zhangchao/riscv_newlib3/riscv --with-arch=rv32gc --with-abi=ilp32
$make newlib

Then, I modify the RISCV and RISCV_prefix in pulp-platform/riscv/tb/core/Makefile as follows:
RISCV                    ?= /home/zhangchao/riscv_newlib3/riscv
RISCV_EXE_PREFIX         = /home/zhangchao/riscv_newlib3/riscv/bin/riscv32-unknown-elf-

However when I compile the hello_world.c in pulp-platform/riscv/tb/core/custom, following errors are shown:
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text+0x10): undefined reference to `sbrk'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-writer.o): In function `_write_r':
writer.c:(.text+0x14): undefined reference to `write'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-closer.o): In function `_close_r':
closer.c:(.text+0x10): undefined reference to `close'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-lseekr.o): In function `_lseek_r':
lseekr.c:(.text+0x14): undefined reference to `lseek'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-readr.o): In function `_read_r':
readr.c:(.text+0x14): undefined reference to `read'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-fstatr.o): In function `_fstat_r':
fstatr.c:(.text+0x12): undefined reference to `fstat'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-isattyr.o): In function `_isatty_r':
isattyr.c:(.text+0x10): undefined reference to `isatty'
collect2: error: ld returned 1 exit status
Makefile:213: recipe for target 'custom/hello_world.elf' failed
make: *** [custom/hello_world.elf] Error 1

It seems that my toolchain lacks something or is installed in a wrong way. So how can I solve this problem.

bluewww, 2019-07-22:
Please provide me with the output of
`make custom/hello_world.elf`
and
`echo $RISCV`

shjdzc12, 2019-07-22:
root@ubuntu:/home/zhangchao/riscv-master-new/tb/core# echo $RISCV
/home/zhangchao/riscv_newlib3/riscv

root@ubuntu:/home/zhangchao/riscv-master-new/tb/core# make custom/hello_world.elf 
/home/zhangchao/riscv_newlib3/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imc -o custom/hello_world.elf -w -Os -g -nostdlib\
	-T custom/link.ld  \
	-static \
	custom/crt0.S \
	custom/hello_world.c custom/syscalls.c custom/vectors.S \
	-I /home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/include \
	-L /home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib \
	-lm -lc 
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text+0x10): undefined reference to `sbrk'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-writer.o): In function `_write_r':
writer.c:(.text+0x14): undefined reference to `write'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-closer.o): In function `_close_r':
closer.c:(.text+0x10): undefined reference to `close'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-lseekr.o): In function `_lseek_r':
lseekr.c:(.text+0x14): undefined reference to `lseek'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-readr.o): In function `_read_r':
readr.c:(.text+0x14): undefined reference to `read'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-fstatr.o): In function `_fstat_r':
fstatr.c:(.text+0x12): undefined reference to `fstat'
/home/zhangchao/riscv_newlib3/riscv/riscv32-unknown-elf/lib/libc.a(lib_a-isattyr.o): In function `_isatty_r':
isattyr.c:(.text+0x10): undefined reference to `isatty'
collect2: error: ld returned 1 exit status
Makefile:213: recipe for target 'custom/hello_world.elf' failed
make: *** [custom/hello_world.elf] Error 1

shangguanzixuan, 2019-07-31:
Maybe this can help you.
https://stackoverflow.com/questions/13235748/linker-error-on-a-c-project-using-eclipse

shangguanzixuan, 2019-08-01:
Sorry，the above method （in the web link） doesn't seem to work

vikramjain236, 2019-08-02:
I ran into similar problem. Is there any solution proposed yet?
Thanks!

shjdzc12, 2019-08-02:
There is still no way to solve the problem. Sorry.发自我的华为手机-------- 原始邮件 --------发件人： vikramjain236 <notifications@github.com>日期： 2019年8月2日周五 晚上10:17收件人： pulp-platform/riscv <riscv@noreply.github.com>抄送： shjdzc12 <shjdzc12@sjtu.edu.cn>, Author <author@noreply.github.com>主    题： Re: [pulp-platform/riscv] unable to generate hello_world.elf (#92)I ran into similar problem. Is there any solution proposed yet?
Thanks!

bluewww, 2019-08-02:
So I looked into this and figured out it is because of different versions of newlib: Before version 2.5  the stubs/symbols are called `sbrk`, `write` etc. while with newer newlib versions they are prefixed with an underscore (`_sbrk`, `_write` etc.). I suppose I will put in some ifdef hack to make it work for both versions.

bluewww, 2019-08-02:
Please try out the PR at #97. If we don't encounter any problems I will merge.

shjdzc12, 2019-08-05:
> I ran into similar problem. Is there any solution proposed yet?
> Thanks!

Please look at the above solution provided by bluewww. I download the new Makefile and custom/systemcalls.c and now run the helloworld successfully.

shjdzc12, 2019-08-05:
> Please try out the PR at #97. If we don't encounter any problems I will merge.

Thanks for your help. I have run the helloworld successfully.

closed 2019-08-05 by bluewww

summerfind, 2024-05-13:
> Please provide me with the output of `make custom/hello_world.elf` and `echo $RISCV`

hello,
I got make not recognized when run your command.

MikeOpenHWGroup, 2024-05-14:
Hi @summerfind, this Issue has been closed for almost five years.  If you have a specific question about the CV32E40P, please review the README and User Manual in detail.  If you still have questions after that, feel free to open a new Issue.