openhwfoundation/cv32e40p #96: Error when running make firmware-vsim-run VSIM_FLAGS=+vcd in riscv/tb/core

vikramjain236, 2019-08-02:
Hi,

When I try to run the riscv tests and compliance tests (Questasim 10.7c), I run into the following issue:

tb_top_vopt -do 'source vsim.tcl; exit -f'
Reading pref.tcl

# 10.7c

# vsim "+vcd" "+firmware=firmware/firmware.hex" -c tb_top_vopt -do "source vsim.tcl; exit -f" 
# Start time: 14:55:41 on Aug 02,2019
# ** Warning: ** Warning: (vsim-7032) The 32-bit glibc RPM does not appear to be installed on this machine.  Calls to gcc may fail.
# 
# ** Warning: ** Warning: (vsim-7032) The 32-bit glibc RPM does not appear to be installed on this machine.  Calls to gcc may fail.
# 
# ** Fatal: ** Fatal: (vsim-3827) Could not compile 'export_tramp.so': cmd = '/bin/gcc   -shared -fPIC -m32  -g  -I. -I"/esat/micas-data/software/Mentor/questasim_10.7c/questasim/include" -I"/esat/micas-data/software/Mentor/questasim_10.7c/questasim/../oem/include" -o "/tmp/vjain@rhea.esat.kuleuven.be_dpi_8440/linux_gcc-4.8.5/export_tramp.so" "/tmp/vjain@rhea.esat.kuleuven.be_dpi_8440/linux_gcc-4.8.5/export_tramp.S"'
# (vsim-50) A call to system(/bin/gcc   -shared -fPIC -m32  -g  -I. -I"/esat/micas-data/software/Mentor/questasim_10.7c/questasim/include" -I"/esat/micas-data/software/Mentor/questasim_10.7c/questasim/../oem/include" -o "/tmp/vjain@rhea.esat.kuleuven.be_dpi_8440/linux_gcc-4.8.5/export_tramp.so" "/tmp/vjain@rhea.esat.kuleuven.be_dpi_8440/linux_gcc-4.8.5/export_tramp.S" >'/tmp/questatmp.r61CS1' 2>&1) returned error code '1'.
# The logfile contains the following messages:
# /bin/ld: cannot find crti.o: No such file or directory
# /bin/ld: cannot find -lc
# /bin/ld: cannot find crtn.o: No such file or directory
# collect2: error: ld returned 1 exit status
# 
# No such file or directory. (errno = ENOENT)
# 
# 
# FATAL ERROR while loading design
# Error loading design
Error loading design
# End time: 14:55:42 on Aug 02,2019, Elapsed time: 0:00:01
# Errors: 1, Warnings: 2
make: *** [vsim-run] Error 12

vikramjain236, 2019-08-02:
Ran the code with vsim flags -64 and the tests runs. However, I get tests failed I_SB_01 and I_SH_01. Closing the issue since the issue with the failed tests are already open.

closed 2019-08-02 by vikramjain236

ramees041, 2021-07-04:
> Ran the code with vsim flags -64 and the tests runs. However, I get tests failed I_SB_01 and I_SH_01. Closing the issue since the issue with the failed tests are already open.

Could you please mention which is the other issue that you were mentioning, because  I am also stuck in similar kind of ssue