openhwfoundation/cv32e40p #100: spike alignment - interrupts should be disabled upon entering ebreak

Itaykeidar, 2019-08-04:
when entering ebreak, spike simulation disables all interrupts, while design does not.

it makes sense that if you are entering an ebreak, then by default you would not want to get any interrupts.
if user would like to enable them back, he can do it manually.

desired behavior should be the same as entering an interrupt.

bluewww, 2019-08-04:
Do you use `PULP_SECURE=0`?

bluewww, 2019-08-04:
Relevant section of `riscv-privileged-v1.10.pdf, p.20`:
```
To support nested traps, each privilege mode x has a two-level stack of interrupt-enable bits and
privilege modes. x PIE holds the value of the interrupt-enable bit active prior to the trap, and
x PP holds the previous privilege mode. The x PP fields can only hold privilege modes up to x, so
MPP is two bits wide, SPP is one bit wide, and UPP is implicitly zero. When a trap is taken from
privilege mode y into privilege mode x, x PIE is set to the value of x IE; x IE is set to 0; and x PP
is set to y.
```
So if we don't set mIE to 0 it is definitely a bug.

Itaykeidar, 2019-08-04:
yes, we are using PULP_SECURE=0.

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB