openhwfoundation/cv32e40p #102: spike alignment - mcause --> debug mode & mstatus mie --> 0

Itaykeidar, 2019-08-04:
In this test register dcsr was set to value 0x9003 in order to enable breakpoints.
After it instruction ebreak is triggered.
 
Here is list of instruction that caused the issue:
 
test_done:
                  li a0, 0x9003
                  csrw dcsr, a0
                  ecall
                  ebreak
 
Spike updates value of mcause and mstatus after execution of ebreak instruction. 
In mstatus register bit mie is set to 0 in order to disable interrupts during debug ROM execution.
Register mcause was updated from 0xb value (ecall) to 0x3 (breakpoint).
In the waveform we can see that these 2 registers were not updated by design after triggering ebreak.

bluewww, 2019-08-04:
Can you please add the waveform image you are referring to?

Itaykeidar, 2019-08-04:
[image]

bluewww, 2019-08-04:
Do you have by any chance a testcase (such as doing something similar to https://github.com/pulp-platform/riscv/tree/master/tb/core/custom) which I can immediately run in the tb? This would really save me some time.

Itaykeidar, 2019-08-04:
[example.zip](https://github.com/pulp-platform/riscv/files/3464671/example.zip)

hope this will help, this is the assembly run

davideschiavone, 2019-08-29:
Any news about it?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB