openhwfoundation/cv32e40p #103: fix some typos in p.clip and p.insert.r

Itaykeidar, 2019-08-04:
o	P.clip* instructions (see attached image named so) all have same opcode. I guess RTL coding (image of RTL) is the correct one?
o	P.insert.r typo – should probably be rs2 and not ls3

also regarding p.insert.r:
We found that most likely the design and spike are behaving as expected and the problem might be the operation description.
We assume the rd bits masking is missing and some typos there so the final formula should be something like  (!(((1<< rs2[9:5]+1) - 1) << rs2[4:0]) & rD) | (rs1[rs2[9:5]:0] << rs2[4:0]) 
where !(((1<< rs2[9:5]+1) - 1) << rs2[4:0]) is a mask.

Also similar for p.insert:
(!(((1<< ls3+1) - 1) << ls2) & rD) | (rs1[Is3:0] << Is2) 

[image]
[image]
[image]

dspg-arch1, 2019-09-18:
Davide, can you please verify that the RTL values above are correct (so we do not have to wait for the spec update)?

davideschiavone, 2019-09-18:
Hello, yes the RTL is correct :) I labeled the bug as documentation as it is a documentation bug.
We are most probably rewriting the documentation in another format, thus the documentation bugs are in line and have less priority than the RTL ones :)

davideschiavone, 2019-12-20:
61593531b7cbb22549e3d18d704b509c450ec412

closed 2019-12-20 by davideschiavone