openhwfoundation/cv32e40p #104: issue with 2 negative values on p.clipur

Itaykeidar, 2019-08-04:
We encountered a problem with p.clipur operation. To show this we prepared a simple test program:
 
_start_jmp:
                  li x3, 0xFFFFFFFF
                  li x4, 0xe9e9e9eb
                   p.clipur x2, x3, x4
 write_tohost:
                                nop
 _exit:
                                j write_tohost
 
According to spec:
[image]

So I see no restrictions to load negative values to the rs1 or rs2 and moreover there is a check if rs1 is negative or zero. 
In our test we are loading two negatives in the x3 and x4 and using them as rs1 and rs2 of p.clipur operation.
Here are the waves:
[image]

test flow:
first pulse of ex_reg_we is the li operation which is loading x3 value.
next two pulses are lui and addi operations to load x4 value.
pulse at TimeA point is the actual p.clipur operation execution.
after this at the next cycle we can see p.clipur result at register x2. according to it's value I think that actually was performed rD = rs2. 
From the description of operation i assume that should be performed rD = 0 as rs1 is negative. We also encountered similar issue with rs1 equal zero.

davideschiavone, 2019-09-23:
Please, find the proposed fix at https://github.com/pulp-platform/riscv/commit/0cc75d1fcfd6c6a7b58014dd7bab903a80f8d90b

Waiting for feedback before merging to the master

yaronbe1, 2019-09-26:
the fix fixed the above bug but introduced a new bug:

When rs1 is 1(0x00000001), rs2 is -370546197(0xE9E9E9EB). result should be -370546197(0xE9E9E9EB) but it's 370546196(0x16161614)  -  wrong

davideschiavone, 2019-09-26:
is it the clipr or clipur?

davideschiavone, 2019-09-26:
it is actually correct in case it is the clipr,

rs1 <= -(rs2 + 1), 1 <= -(-370546197+1) which is 370546196, so the result should be 0x16161616.

This comes from the fact that negB + 1 = -B, so the clip returns negB which is -B -1, i.e. -(B+1)

It that is the clipur , then it's a bug because it should return 0xE9E9E9EB.

yaronbe1, 2019-09-27:
I am attaching the waves. 
please look at the time marker, AFAIK, the core is processing a clipur instruction:

[image]

davideschiavone, 2019-09-27:
yes Yaron you are righ, my bad.

this should fix it: 
https://github.com/pulp-platform/riscv/commit/8eec2b452c7ca1a7cf70970f0efb8c3c6005486f

Let me know whether this solves the issues, if so I merge to the master.

thanks
Davide

yaronbe1, 2019-10-02:
still not robust enough:
clipur rs1 = 0, rs2 = ebebebe9  should give rd = 0 but real result rd = ebebebe9

davideschiavone, 2019-10-07:
yet another tentative fix: https://github.com/pulp-platform/riscv/commit/53da0833e7b610852bc67db62d34f64ad4c5e32c

yaronbe1, 2019-10-08:
no issues after this fix
you can commit

davideschiavone, 2019-10-08:
Thanks a lot

commit 8eec2b45 (2019-09-27): fix issue introduced to fix #104
closed 2019-10-08 by davideschiavone