openhwfoundation/cv32e40p #105: issue with pv.dotup.sci.h instruction

Itaykeidar, 2019-08-04:
Source code of example:
                   li x2, 0x0
                  li x3, 0x000FFFF
                   pv.dotup.sci.h x2, x3, 0x38
 
 Error message in the simulation results:
UVM_ERROR : ... [SPIKE_DBG] EX stage. Register #2 aka sp update mismatch. Expected value is 0x37ffc8, actual value is 0xfff70008

[image]

Predicted result following the formula should be: 0xFFFF*0x38 + 0x0*0x38 = 37FFC8
ri5cy during simulation set's value of the register x2 =  0xfff70008 as the result of this command execution

davideschiavone, 2019-09-11:
Hello, we tried to solve it here: 520ef0a418e3dc88d760fcb7092cdb15a81e724d

Can you try it please and report whether it solves the issue?

thanks
Davide

davideschiavone, 2019-09-17:
It has been confirmed to be fixed

closed 2019-09-17 by davideschiavone