openhwfoundation/cv32e40p #106: pv.sdotup.sci.b instruction issue

Itaykeidar, 2019-08-04:
Error is generated from  command pv.sdotup.sci.b a2, t2, 0x3e 

[image]

Disasm
 
•	the first bag is from disasm. It transform 62 -> 0x30 instead 0x3e.
•	The second
According documentation and waveform:
pv.sdotup.sci.b rD, rs1, Imm6
t2 = x7  = 0x00_80_00_00
rd = x12 = 0x00_80_00_00
Imm6 = 0x3E
so
rd = rd + (rs1[0]*Imm6 +  rs1[1]*Imm6 + rs1[2]*Imm6 + rs1[3]*Imm6) = 
= 0x00_80_00_00 + (00*0x3E + 0x80*0x3E+ 00*0x3E+ 00*0x3E) = 0x00_80_00_00 + 0x00_00_1F_00 = 0x00_80_1F_00
According the documentation all operations are unsigned.
 
But in design results 0x00_80_7F_00.
I looked through the design. 
1.	It uses signed Imm6: 0xFE instead 0x3e;  0x80*0xFE= 0x7F00
[image]

Question is: 
1.	Is it  a bug of documentation and Imm6 should be considered as signed number?
2.	Or Is it a bug of decoding of the command and  imm_b_mux_sel must be 7 instead 6, or is it race of signal?

[image]

davideschiavone, 2019-09-11:
Hello, we tried to solve it here: 520ef0a418e3dc88d760fcb7092cdb15a81e724d

Can you try it please and report whether it solves the issue?

thanks
Davide

davideschiavone, 2019-09-17:
It has been confirmed to be fixed

closed 2019-09-17 by davideschiavone