openhwfoundation/cv32e40p #107: IRQ arrive on the last instruction of HW loop

Itaykeidar, 2019-08-04:
In the waveform we can see next coincidence: hardware loop #1 is enabled, number of iterations is 6, the pointer to the last instruction in the loop is 0x8000_065E.
Instruction in the pointer 0x8000_065C executed normally, but before execution of instruction in pointer 0x8000_065E, interrupt arrived.
After exiting from interrupt processing function CPU finished loop iteration and moved to pointer 0x8000_064C. (pointer 0x8000_065C is the beginning of the loop).

Screenshot of waveform:
[image]

So we can conclude that last instruction in the loop was interrupted by IRQ processing and after it  not executed.

To perform jump in hardware loop execution next signal is used: hwlp_jump_o of the riscv_hwloop_controller module.
This signal takes into account account next PC after execution of current instruction and if next PC is equal to HW loop end address hwlp_jump_o becomes 1.
In the next wavefrom we can see that hwlp_jump_o is active on ID stage of  0x8000_064C instruction.
Next time signal hwlp_jump_o was active for a short time during mret instruction execution.
It looks like design  mixed up execution of mret with the last instruction in the loop.

[image]

Itaykeidar, 2019-08-28:
Hi,

it seems the problem is the prefetch of next instruction.
here is a more simple example:
[image]

on the highlited bottom, is the select between the last fetched instruction and the next.
this chooses between the last HW loop command and the first command.
wrongfully it chooses the later.

this happens due to an interrupt that happened on the last hw line "0x800002fA":
[image]

it happens due to wrong selection of the address:
[image]

probably due to no clear in state machine, or missing condition.

thanks,
Itay K.

davideschiavone, 2019-08-28:
Thanks a lot Itay!

I will try to reproduce it and fix it as soon as possible, this issue has the top priority to me

davideschiavone, 2019-09-19:
Hello,

can you please send me the piece of assembly that you used?

or the binary of the HW loop part so that I can replicate it? I need to replicate the bug with the same compressed instructions that you are using.

Thanks
Davide

davideschiavone, 2019-09-19:
I confirm it is a bug, can you please check with this fix?
https://github.com/pulp-platform/riscv/commit/78c4c448bdc7cb0094c2d759ca5352236e7e44c6

yaronbe1, 2019-10-10:
the fix for #148 fixes this issue also

davideschiavone, 2019-10-11:
Fixed by https://github.com/pulp-platform/riscv/commit/f19f0722b06b53a59cb38ea982d909f2d8797d13

davideschiavone, 2019-11-05:
The solution proposed contains a combinational loop,
please find the proposed fix for that combinatorial loop here:
https://github.com/pulp-platform/riscv/issues/190

I will merge to master once I get feedback from your functional verification and sorry for inconvenience

davideschiavone, 2019-11-12:
is also this one fixed? @yaronbe1

yaronbe1, 2019-11-12:
> External Email - Use caution when responding, clicking and/or downloading attachments.

Yes

DSP Group, Inc. automatically scans all emails and attachments using MessageLabs Email Security System.

DSP Group, Inc. automatically scans all emails and attachments using MessageLabs Email Security System.

closed 2019-11-12 by davideschiavone