openhwfoundation/cv32e40p #108: handling order and priority of irq/ecall/ebreak/etc...

Itaykeidar, 2019-08-04:
the assumption is that ordering of events is :
branch-debug-irq-illegal-rest

it seems is not just like that, or we misunderstood something.
after some investigations wefound that when irq request appears in will be acknowledged only at the next is_valid ==1 time.
So whenever irq request appear one more instruction will be handled first. For example:

[image]

and the  asm part:

...
80000c24:
09b1      addi
s3,s3,12
80000c26:
0231d6b3 divu
a3,gp,gp
80000c2a:
ef3f6197 auipc
gp,0xef3f6
80000c2e:
1d95      addi
s11,s11,-27
80000c30:
8f4d      or
a4,a4,a1
... 

At the baseline interrupt request signal was set. Meanwhile the divu operation was at the EX stage.
At the next is_valid signal irq_flush state was set as a next state but the handled operation was auipc.
And only the next operation handled is irq.
can the following be answered/added to spec?
•	Irq handling is happening one operation after the irq request is set, right?
•	Is this behavior is constant or there are special conditions that break this algorithm?
•	What will be the behavior when the two b2b irq request will be set?
•	Priority is: branch-irq-ebreak-illegal-rest, right?

dspg-arch1, 2019-09-18:
There seems to be another cornercase/ priority issue situation: 
IRQ was activated when wfi instruction had been scheduled for execution.
Because IRQ has higher priority rather than regular instruction, execution of wfi instruction was terminated. That is, we expected to see value of "mepc" register set to  the address of the wfi instruction, but instead we saw it set to the following instruction

Is this the correct behavior? Where should it be specified?

davideschiavone, 2019-09-19:
Hello @dspg-arch1 ,

the priority is:

taken branch
data error
fetch error

then if the instruction is valid (so if there is an instruction in the ID stage, which means if the IF stage finished fetching)
debug
interrupt
illegal
rest

if there is a load or division or any instrction that is taking a lot to be executed in the EX stage, the first instruction of the ISR is waiting for it to be finished before begin executed.

If the WFI has been executed, the interrupt is taken when the core is in the SLEEP state, so the WFI has been executed and thus the next PC is saved into the MEPC as the WFI has been already executed.

If the interrupt is taken before the WFI (i.e. when the controller is in the DECODE stage rather than SLEEP), the MEPC is the WFI one as the WFI hasn't been executed yet.
Is what I am writing what you are seeing?

yaronbe1, 2019-10-02:
Hi Davide

the following is feedback from our verification team:

We reviewed response from Davide about cornercase situation when IRQ raised simultaneously with WFI instruction execution.

"If the WFI has been executed, the interrupt is taken when the core is in the SLEEP state, so the WFI has been executed and thus the next PC is saved into the MEPC as the WFI has been already executed.

If the interrupt is taken before the WFI (i.e. when the controller is in the DECODE stage rather than SLEEP), the MEPC is the WFI one as the WFI hasn't been executed yet."

From these 2 paragraphs we can not determine in which case WFI instruction is skipped using MEPC register. Can you please show in the waveform which signals need to check in order to understand which WFI instruction will be executed and in which case it would be skipped?

Let's review case when WFI execution was interrupted by IRQ and MEPC register contain value of WPi instruction pointer:
[image]

Case when IRQ is simultaneous with WFI, but value of MEPC is PC+4.
[image]

WFI instruction is skipped because id_valid & is_decoding is equal to 1'b0 when PC is 0x800003A2 therefore ID stage of WFI instruction was not executed. 
Because MEPC is PC + 4, after returning from IRQ handler ri5cy starts executing of instruction next after WFI, while in the first situation WFI was excuted after IRQ.

How can we know in advance difference between these 2 cases from TB perspective?

for DSPG internal: To open simulation results use next command:
lbrun -C simvision -p high -- simvision -64bit -waves /home/dtmp/run/dbm10l/denisg/ri5cy/ws_dbm10l_test/wfi_issue/vlg_sim/waves.shm

Itaykeidar, 2019-10-23:
Hi Davide, any comment on the above?
would be really helpful.

thanks,
Itay K.

Silabs-ArjanB, 2020-06-15:
Hi @Itaykeidar We do not really want to specify when an interrupt is taken based on some internal signals of CV32E40P. Interrupts are inherently not precise in relationship to the instruction stream (unlike synchronous exceptions) and therefore it does not make sense for us to specify this behavior as it might change in future RTL updates (e.g. due to registering incoming interrupts to break some critical path). If an interrupt occurs close to a WFI, it might happen before or after the WFI executes, there is not much more that can be said about this (you can find the relevant signals in the RTL if you really want, but this would only be helpful for some artificial testbench, not for actual software as that software cannot look at any such internal signals). If the system software depends on this ordering, then I would question the software architecture.

The priority of the precise/synchronous exceptions has been answered. Could you please close this issue if this answers your question? (We will consider the issue resolved automatically after 7 days unless we receive follow up questions from you)

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it has been answered at least 7 days ago. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions. Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB