openhwfoundation/cv32e40p #110: execution of pv.abs even if op code bits [25:20] are not zero

yaronbe1, 2019-08-04:
In the ri5cy spec op code bits [25:20] of command pv.abs are defined as zero.
We have found that the ri5cy executes the command even if those bits are not zero, I.E. opcode  0x71169557  or opcode 0x71169557. 

The code 0x71169557 for instance is illegal for RI5CY.  This code is similar to pv,abs command, but it differs from pv.abs  by 
cmd[26:20] not being zero.
RD register is 01010 - x10.
The CPU doesn't understand that the command is illegal and executes the command, changing register x10 to 0.
 
The same situation is for 0x71a208d7 code.

maybe the intention was that RS2 field in this command is a don’t care. in this case the spec should be updated.

davideschiavone, 2019-08-05:
Yes this is actually a DON'T CARE. We are going to change the Documentation

dspg-arch1, 2019-09-04:
Just to clarify
The all zeros "don't care" field is 24:20 , bit 25 is part of the instruction decode, it is "care"
-abg DSPG

davideschiavone, 2019-12-20:
61593531b7cbb22549e3d18d704b509c450ec412

closed 2019-12-20 by davideschiavone