openhwfoundation/cv32e40p #99: align design to spike simulator on setting mpie to 1'b0 on mret

Itaykeidar, 2019-08-04:
We encountered a difference between spike and ri5cy behavior regarding the value of mpie register once MRET or DRET are executed:
During DRET the spike loads the mpie value into mie reg (as it should as mentioned in the ri5cy spec) and also clears the mpie bit (loads zero).
The ri5cy design loads the mpie value into mie reg (as it should as mentioned in the ri5cy spec) but also sets mpie to one (if I understand it correctly I would expect it to clear it to zero, as this is also it’s reset value).
 
 
From cs_registers module line ~830:
 
      csr_restore_mret_i: begin //MRET
        mstatus_n.mie  = mstatus_q.mpie;
        priv_lvl_n     = PRIV_LVL_M;
        mstatus_n.mpie = 1'b1;
        mstatus_n.mpp  = PRIV_LVL_M;
      end //csr_restore_mret_i
      csr_restore_dret_i: begin //DRET
        mstatus_n.mie  = mstatus_q.mpie;
        priv_lvl_n     = PRIV_LVL_M;
        mstatus_n.mpie = 1'b1;
        mstatus_n.mpp  = PRIV_LVL_M;
      end //csr_restore_dret_i

it makes sense to align to spike simulator, as it aligns with the reset value of the ff as well.

bluewww, 2019-08-04:
Are you sure about setting `mpie` to 0? This would contradict the privilege spec
This is what `riscv-privileged-v1.10.pdf` says:
```
The MRET, SRET, or URET instructions are used to return from traps in M-mode, S-mode, or
U-mode respectively. When executing an xRET instruction, supposing x PP holds the value y, x IE
is set to x PIE; the privilege mode is changed to y; x PIE is set to 1; and x PP is set to U (or M if
user-mode is not supported).
```
Can you maybe point me to the code in spike where that happens?

Itaykeidar, 2019-08-04:
Hi,
please see that you are quoting a statement about behavior of a move from one privilege mode to another:
"...the privilege mode is changed to y..."
of course in this case logic should be set to 1'b1.
when there is no privilege mode to another, basically the spec does not describe what the behavior should be on normal mode, so i imagine it is a "don't care", but it makes sense to align with spike and with FF reset value.

spike code:
https://github.com/riscv/riscv-isa-sim/blob/88a852836acb4c7166b1aa4102e11354bfd99234/riscv/processor.cc
See lines 395

} else {
    reg_t vector = (state.mtvec & 1) && interrupt ? 4*bit : 0;
    state.pc = (state.mtvec & ~(reg_t)1) + vector;
    state.mepc = epc;
    state.mcause = t.cause();
    state.mtval = t.get_tval();

    reg_t s = state.mstatus;
    s = set_field(s, MSTATUS_MPIE, get_field(s, MSTATUS_MIE));
    s = set_field(s, MSTATUS_MPP, state.prv);
    **s = set_field(s, MSTATUS_MIE, 0);**
    set_csr(CSR_MSTATUS, s);
    set_privilege(PRV_M);

bluewww, 2019-08-04:
You seemed to have shown me the code where the the trap is being taken (`
void processor_t::take_trap(trap_t& t, reg_t epc)`) not left. I looked around a bit and I found [this](https://github.com/riscv/riscv-isa-sim/blob/88a852836acb4c7166b1aa4102e11354bfd99234/riscv/insns/mret.h) but it seems to assume there is more than one privilege mode.

Anyway we don't mind aligning with spike in such cases.

davideschiavone, 2019-08-04:
This is indeed a bit strange. 
 @Itaykeidar  and I had the discussion about it. 
We assumed that when you move from priviledge X to Y, X and Y could be the same (machine to machine).
But apparently Spike does differently.
So as it is not staten anywhere, the MPIE in this case is a DONT CARE.

So let's align to Spike in my opinion. @haugoug what do you think? How is it done in your simulator?

Thanks
Davide

silabs-PaulZ, 2020-07-13:
Hi @Itaykeidar ,

We are currently using the Imperas model which is matching the behavior where MPIE=1 after MRET. I'm having some trouble getting the Spike simulator running. Could you see if that behavior has changed? My understanding is the same as @davideschiavone and @bluewww in that X to Y can be the same privilege mode. I would be in favor of not modifying the RTL and perhaps post it as an issue for Spike (as the core is seemingly aligned to the specification).

Silabs-ArjanB, 2020-07-23:
Hi @Itaykeidar We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-23 by Silabs-ArjanB