openhwfoundation/cv32e40p #112: errors when running coremark on the core

shjdzc12, 2019-08-05:
When I run helloworld.c, the platform is OK and I get "helloworld!" on the screen.
However, when I replace helloworld with coremark. It will have the following errors:

"out of bounds read from 970a8313"

I find out that this error report comes from the file "riscv/tb/core/mm_ram.sv", I copy the referenced code here for you:
*******
always_comb begin: read_mux
        data_rdata_o = '0;
        if(select_rdata_q == RAM) begin
            data_rdata_o = ram_data_rdata;
        end else if (select_rdata_q == ERR) begin
            $display("out of bounds read from %08x", data_addr_i);
        end
end

if (data_addr_i < 2 ** RAM_ADDR_WIDTH) begin
                    select_rdata_d = RAM;
                    ram_data_req = data_req_i;
                    ram_data_addr = data_addr_i[RAM_ADDR_WIDTH-1:0];
                    ram_data_wdata = data_wdata_i;
                    ram_data_we = data_we_i;
                    ram_data_be = data_be_i;
                end else select_rdata_d = ERR;
******

So in my opinion, this error comes from the fact that actual data address width is bigger than RAM_ADDR_WIDTH.
Actual data address width is 32(according to the print value 970a8313). However, In tb_top, RAM_ADDR_WIDTH is set to be 22, which is much smaller than 32. When I change the RAM_ADDR_WIDTH to be 32, another problem happens. Here are the errors reported:

12794835ns: Illegal instruction (core 0) at PC 0x00000000:
"../../rtl/riscv_id_stage.sv", 1625: tb_top.riscv_wrapper_i.riscv_core_i.id_stage_i.unnamed$$_19: started at 12794840000ps failed at 12794840000ps
	Offending '(!$isunknown(instr_rdata_i))'
Instruction is valid, but has at least one X

The instruction can't be even decoded correctly. So How can I run the coremark on this core?

shjdzc12, 2019-08-05:
Are there some ways to support the instruction whose width is bigger than 22?

bluewww, 2019-08-05:
The RAM_ADDR_WIDTH parameter just defines how big the memory is. In that case it is 2 the power of 22 bytes which should be more than enough to run it. The linker script ensures that code and data is placed into that address range. So the issue is somewhere else. You can look at the instruction trace or the waveforms to figure out the issue. The error you have given normally happens when you execute garbage code.

Silabs-ArjanB, 2020-09-24:
Hi @shjdzc12 Is this issue still present? If so, can you please provide a way to reconstruct this issue (full program code, linker file, etc.)?

Silabs-ArjanB, 2020-10-28:
Hi @shjdzc12 We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-10-28 by Silabs-ArjanB