openhwfoundation/cv32e40p #116: HINTS implemented?

yaronbe1, 2019-08-06:
We got an illegal instruction exception for the following command: c.add  zero,a2

as far as I understand, the riscv spec states that c.add with  zero as destination should be treated as a HINT.
From riscv spec page  106:
C.ADD adds the values in registers rd and rs2 and writes the result  to register rd. C.ADD expandsinto add rd, rd, rs2. C.ADD is only valid when  rs2̸=x0; the code points with rs2=x0 correspondto the C.JALR and C.EBREAK  instructions. The code points with rs2̸=x0 and rd=x0 are  HINTs.

According to the riscv spec a HINT should be treated by the  core as a NOP and not as an illegal instruction.

As I mentioned above the  ri5cy core gave an illegal instruction exception on this command. 

HINTs not implemented in the ri5cy?

davideschiavone, 2019-08-06:
Hi,

indeed they are not implemented, in your specific case the illegal trigger comes from here:
https://github.com/pulp-platform/riscv/blob/master/rtl/riscv_compressed_decoder.sv#L258

Are they mandatory?

if so, this is a bug. Anyway thanks for reporting it :)

Davide

Silabs-ArjanB, 2020-06-08:
Hi @yaronbe1 this issue has now been fixed. Can you please close it if you agree it has been fixed? The issue will automatically be closed in 7 days otherwise.

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB