openhwfoundation/cv32e40p #118: Post-synthesis simulation with helloworld custom program does not work

vikramjain236, 2019-08-15:
Hi,
I am trying to re-run the helloworld code from tb/core/custom/ with a synthesized netlist in X technology. For that, I generated the verilog netlist in Cadence genus and in the makefile under tb/core I made changes to include the verilog netlist and the technology verilog codes as the RTLSRC. When I run the simulation, the program does not run and looking at the GUI a lot of the signals are pulled to XXXX for some reason. Any help would be appreciated:

[image]
[image]
[image]

vikramjain236, 2019-08-28:
Hi,
Can anyone help me with this?

bluewww, 2019-08-28:
Well, the testbench is not meant for synthesis. Furthermore we can't really tell you why it happens since we can't reproduce your issue.

vikramjain236, 2019-08-28:
Do you have any testbench that you use to verify the post-synthesis simulations?

closed 2019-08-29 by vikramjain236

bluewww, 2019-08-29:
Could you resolve your issue?

vikramjain236, 2019-08-29:
Yes, there was this debuq_req_i port which was the problem. Since the debug_req_i signal comes from the riscv_wrapper.sv and I was only synthesizing the riscv_core, the debug_req_i input port to the riscv_core was unconnected. In the riscv_controller there are some if conditions which test debug_req_i and if the condition is unconnected the halt_if_o and halt_id_o are set to XXXX and the whole simulation breaks. So, I removed the debug_req_i from the if conditions.

ranaya123, 2019-09-21:
@vikramjain236 : Hi, which platform and which specific core you were trying to simulate in this case ? I ran into a similar issue while performing gate-level simulations on RI5CY core of Pulpissimo platform. This is further explained in following issue page:
https://github.com/pulp-platform/pulpissimo/issues/77

Apart from the debug_req_i, did you have to resolve some other issues in ID stage ?

Thanks in advance

vikramjain236, 2019-09-23:
@ranaya123 Hi, I was performing synthesis only on the RISCY core. You have to make sure that in the FSM, all the condition checks for debug_req_i is properly handled. Also, the debug_mode_q signal is an issue which you should take care of. I did not face any issue in the ID stage, most of the problem for me was the FSM in the riscv_controller.

ranaya123, 2019-09-23:
@vikramjain236 : Hi, ohh so you just ran the test-bench in :
https://github.com/pulp-platform/riscv/tree/master/tb
not the pulpissimo test-bench ?

vikramjain236, 2019-09-23:
Correct.